gnuradio/gnuradio #4495: GNU Radio Companion - Constellation Modulator - unexpected Time Sink plots

KorgBoy, 2021-04-06:
Using a Constellation Object block with constellation points set as  [-1-1j, 1-1j, 1+1j, -1+1j], and using a symbol map set as [0, 1, 2, 3] ----- I execute the project file (contained in my attached zip file) in GNU Radio Companion 3.8, and notice signal (waveform) plots from the QT GUI Time Sink block not being what I expect to see. It could be an issue. Not sure!

I have attached a screenshot of my GNU Radio Companion block diagram, and a screenshot of one of the Time Sink plots --- for the case where I set the Vector Source value as '3'.

[image]

[image]

[4QAM_1.zip](https://github.com/gnuradio/gnuradio/files/6261131/4QAM_1.zip)

willcode, 2021-04-06:
The vector is packed, so `[3]` is `00 00 00 11`. Placing a Unpacked to Packed block right after the Vector Source will give you a constant symbol, as will `[0xff]` (`11 11 11 11`).

KorgBoy, 2021-04-06:
willcode - that was awesome. Thanks so much!

That helped an incredible lot. The details about the vector being 'packed' is so important here. Getting the expected results now, after trying what you mentioned.

When I put in a Vector Source number of '255' (ie. 11 11 11 11, which is four lots of '3'), the waveform plot from the Time Sink block showed Signal 1 being -0.707, and Signal 2 being +0.707 ...... which corresponds nicely with the constellation point -1 + 1j.

And when I put in a Vector Source value of '85' (ie. 01 01 01 01, or four lots of '1'), the Time Sink plot had Signal 1 being +0.707 and Signal 2 being -0.707, which coincides with the constellation point +1 - 1j. 

The Time Sink plot for Vector Source value '85' is shown in the screenshot below (worked nicely) :
[image]

I have added zipperd .grc files for the case where I used Source value '85' alone, and also for the case with Source value '3' (with an Unpacked to Packed block set to 2 bits per chunk). Both working excellently. 

Thanks again willcode. I wouldn't have known what was going on without your help here. Greatly appreciated.

[4QAM_2.zip](https://github.com/gnuradio/gnuradio/files/6262406/4QAM_2.zip)

[4QAM_3.zip](https://github.com/gnuradio/gnuradio/files/6262455/4QAM_3.zip)