gnuradio/gnuradio #4500: Does gnuradio have a plan to support i18n?

taotieren, 2021-04-06:
Does gnuradio have a plan to support i18n?

dkozel, 2021-04-06:
Hello @taotieren

Translation of GNU Radio and GNU Radio Companion are being discussed here:
https://github.com/gnuradio/gnuradio/issues/4223

There is unicode support in the latest release, but there's only been limited testing of it actually using non Roman characters. No one that I know of has looked at Internationalization issues such as date and number formats or writing direction.

Is there a particular part of Internationalization that you're interested in? Would you be interested in helping implement some of it?

marcusmueller, 2021-05-14:
Closing this as this issue is just a question, no action has been proposed and we've not heard from you in a month, sadly :(

closed 2021-05-14 by marcusmueller