gnuradio/gnuradio #4503: Learn about GNU Radio users

ei8fdb, 2021-04-06:
## The "UX" of Gnu Radio

I'd like to propose some UX work to learn about GNU Radio users, specifically GNU Radio Companion users.

### Why learn about GRC users?

In order to improve the UX of software, we need to understand who those users are, what they use the software for, their pain points with using the software (maybe software bugs, maybe accessibility issues, maybe knowledge gaps, etc). Once we about our users we can improve the experience they have with the software. [UX without research is not UX](https://www.nngroup.com/articles/ux-without-user-research/).

Learning about our users also makes it easier to develop things in the right way - we can test what we're going to build with users before we've written any code. Saves time and precious electrons!

### How do we learn about our users?

There are lots of different methods of learning about users - surveys, interviews, usability tests of UI mock-ups, and lots more. Each of these can help us understand different aspects of our users, their pain points, what they use GRC for, what they want to use GRC for but can't, etc. These methods will provide us with a mix of qualitative and quantitative data from users.

### OK I've read this far and don't hate the idea, what do you want to do?

Thanks for sticking with me this far.

#### Step 1. Call for involvement

The first thing I propose is forming a group of users who are interested in taking part in these UX sessions. The way to do this is to advertise this opportunity ("Call for involvement") to **as diverse and wide as possible a group of people** who use GRC (those who are and aren't members of the GNU Radio community). It could look something like [this blogpost](https://www.ei8fdb.org/thoughts/2020/03/pip-ux-study-recruitment/). This blogpost explains the objective, what the person gets out of it, how we use the information we learn.

It could be a blogpost, a page on the GR wiki, etc. It would be advertised to the mailing list, social media, sent to people we know are GR users - academica, industry, ham radio communities, others I don't know about, at GRCon, etc.

Advertising this call as widely as possible is critical to get as diverse, and inclusive as possible involvement.

The person who is interested completes a short sign-up form - providing some basic information about themselves, how they use GRC, what they use it for, their contact details, etc.

#### Step 2. Learn about our users

Once we have an acceptable number (more than 10 less than 1,000,000 :)) of people signed up, we can then start learning about our users.

We can ask for their input in user related decisions - for example it'd be useful to understand the different native/first languages of our users so we can better decide what to do about [internationlisation and localisation](https://github.com/gnuradio/gnuradio/issues/4223), also to understand the difficulties users with visual disabilities or colour deficiencies have when using GRC, also be useful to do usability testing of new designs, like [this one](https://github.com/gnuradio/gnuradio/issues/4330) or [this one](https://github.com/gnuradio/gnuradio/issues/4100) or fixing UI bugs like [this one](https://github.com/gnuradio/gnuradio/issues/3423).

### Comment wanted

Before I start putting this together, I'd like to get some feedback from others. Are there any questions maintainers have about users - is there anything you'd like to understand about them? Post here, or you can email me - bernard+gnuradioATei8fdbDOTorg.

duggabe, 2021-04-06:
Great! A related issue is the documentation for the GRC blocks. In GR 3.9+ there is a link in the block Properties / Documentation tab which takes the user to the Wiki page for that block. Is it useful? Is that sufficient? What more could be done?

777arc, 2021-04-06:
Yeah I always thought it would be cool to have a mini-browser window open the wiki page for the block, like when you click the documentation tab, perhaps in a way that renders the page without the sidebar and top bar.  I've seen some other software do that, they try to hide that its actually a web browser but its usually pretty obvious.

ei8fdb, 2021-04-07:
> Great! A related issue is the documentation for the GRC blocks. In GR 3.9+ there is a link in the block Properties / Documentation tab which takes the user to the Wiki page for that block. Is it useful? Is that sufficient? What more could be done?

Thanks @duggabe, how uses use documentation, why, for what, etc  is a good topic to understand. I'll add that to the list.

ei8fdb, 2021-04-07:
> Yeah I always thought it would be cool to have a mini-browser window open the wiki page for the block, like when you click the documentation tab, perhaps in a way that renders the page without the [...]

Thanks @777arc - do you have any examples of software that do that? Or a screenshot?

ei8fdb, 2021-04-10:
Updated [ticket above](https://github.com/gnuradio/gnuradio/issues/4503#issue-851784630) with the call for involvement and the UX Study sign-up form, listed under "Steps".

NeuroForLunch, 2021-06-19:
I'm already working on improving the documentation tab of the blocks because clicking the link to wiki page and seeing pretty much nothing more than is already shown via tooltips is kinda useless. And if there is new information, why not include that info in the tab instead of having to open the browser? If there was a way to include hyperlinks and images in the descriptions then the wiki data could be transfered to the documentation tab increasing workflow speeds by a lot.

[image]

NeuroForLunch, 2021-06-19:
Each block should be fully explained so a new user can click on the Documentation and understand what it does and make sure to include any definitions for words that are not common English. Acronym usage is also a problem unless it is defined beforehand. A person who knows what the block does isn't going to read the Documentation and a person who sorta knows will skim over the parts they already know, while a person who doesn't know what a block does should learn how it can be used in their project. There's several hundred blocks and plenty of example flows but the blocks are missing documentation about what exactly they're doing.

777arc, 2021-06-19:
> Each block should be fully explained so a new user can click on the Documentation and understand what it does and make sure to include any definitions for words that are not common English. Acronym [...]

Yeah I mean you pretty much described our goal, although it's unfair to say "the blocks are missing documentation about what exactly they're doing", I'm looking at plenty of pages now that explain exactly what the block does.  Plenty of room for improvement though.

777arc, 2021-06-19:
As far as rendering the wiki page within the documentation tab, seems like low hanging fruit is using a mini-browser window for the tab, and then we can also try to figure out a way to export the wiki content and get it displayed nicely, but it's not easy, we need the pics and equations and stuff...

NeuroForLunch, 2021-06-19:
I agree that some blocks have good explanations but there's hundreds of blocks and not all have good explanations. For instance take the CTCSS Squelch block. It doesn't even define what CTCSS is: https://wiki.gnuradio.org/index.php/CTCSS_Squelch

Wikipedia does a better job: https://en.wikipedia.org/wiki/Continuous_Tone-Coded_Squelch_System

I'm sure theres even better explanations elsewhere so why not use the explanations from experts and just reference them?

Having images render would definitely be a huge plus so users wouldn't have to visit an external website. I don't know how to incorporate this but since the tab is using GTK why not use https://developer.gnome.org/gtk3/stable/GtkImage.html#gtk-image-set-from-file ?

777arc, 2021-06-19:
The GRC rewrite is using pyqt so that should make adding this stuff easier.  You're very correct that plenty of blocks are missing important docs, we can definitely use help filling them out, there are just so many it's difficult for one or two people to do them all.

NeuroForLunch, 2021-06-19:
> The GRC rewrite is using pyqt so that should make adding this stuff easier. You're very correct that plenty of blocks are missing important docs, we can definitely use help filling them out, there are just so many it's difficult for one or two people to do them all.

I just started a discussion about this and saw your post seconds later. https://github.com/gnuradio/gnuradio/discussions/4790

jaredd, 2021-07-09:
Some blocks in the wiki have example GRC flowgraph images, but it seems like it should be possible to link to example flowgraphs that come with the module. Clicking the link could bring up the flowgraph in GRC.