gnuradio/gnuradio #4505: Default embeded python code block doesn't work when parameters are used to specify input/output sizes

tclarke, 2021-04-07:
Example:
...
def __init__(self, vector_size=1):
    gr.basic_block.__init__(self, name="foo", in_sig=[(np.float32, vector_size)], out_sig=[])
    self.vector_size=vector_size
...
This will use the default vector_size=1 when GRC calculates input sizes. The self.vector_size=vector_size needs to happen before the base class __init__ is called.

dave-pi, 2021-04-09:
Just to clear what happens because it can be confusing. When you are designing the flowgraph, the only thing that matters is the value of the length in the epy block construtor. This can come from a literal hardcoded value, a parameter, an instance attribute, a local variable, whatever you want as long as the value passed in `in_sig` is correct when the constructor is executed (which happens when you press OK in the dialog box or connect arrows). These are all valid:
```
def __init__(self, example_param=1.0, foo=9999):  # only default arguments here
        """arguments to this function show up as parameters in GRC"""
        local_foo = 9999
        self.foo = 9999
        gr.basic_block.__init__(
            self,
            name='Embedded Python Block',   # will show up in GRC
            in_sig=[(np.complex64, 1024)],
            out_sig=[np.complex64]
        )
```
```
def __init__(self, example_param=1.0, foo=1024):  # only default arguments here
        """arguments to this function show up as parameters in GRC"""
        local_foo = 9999
        self.foo = 9999
        gr.basic_block.__init__(
            self,
            name='Embedded Python Block',   # will show up in GRC
            in_sig=[(np.complex64, foo)],
            out_sig=[np.complex64]
        )
```
```
def __init__(self, example_param=1.0, foo=9999):  # only default arguments here
        """arguments to this function show up as parameters in GRC"""
        local_foo = 1024
        self.foo = 9999
        gr.basic_block.__init__(
            self,
            name='Embedded Python Block',   # will show up in GRC
            in_sig=[(np.complex64, local_foo)],
            out_sig=[np.complex64]
        )
```
```
def __init__(self, example_param=1.0, foo=1024):  # only default arguments here
        """arguments to this function show up as parameters in GRC"""
        local_foo = 9999
        self.foo = 1024
        gr.basic_block.__init__(
            self,
            name='Embedded Python Block',   # will show up in GRC
            in_sig=[(np.complex64, self.foo)],
            out_sig=[np.complex64]
        )
```
```
def __init__(self, example_param=1.0, foo=1024):  # only default arguments here
        """arguments to this function show up as parameters in GRC"""
        local_foo = 9999
        self.foo = foo
        gr.basic_block.__init__(
            self,
            name='Embedded Python Block',   # will show up in GRC
            in_sig=[(np.complex64, self.foo)],
            out_sig=[np.complex64]
        )
```

A this point it doesn't matter what you put in the parameter-editing text box in the dialog box. The only thing that matters is what is inside the `epy_block_0.py` file.
The value in the textbox is only evaluated when the flowgraph is run. At that point, the block is instantiated and connected, and this time the value you put in the text box is passed as the argument. Have a look inside the main python file that is generated:
```
self.epy_block_0 = epy_block_0.blk(example_param=variable_0, foo=<value_from_textbox>)
...
self.connect((self.epy_block_0, 0), (self.blocks_throttle_0, 0))
```

The callback thingy refers to the fact that if you set the parameter textbox to a variable and the epy block has an instance attribute with the same name as the parameter then whenever the variable is set, it will also update the instance attribute:
```
def set_variable_0(self, variable_0):
    self.variable_0 = variable_0
    ...
    self.epy_block_0.foo = self.variable_0
```
But this is pointless for that use case, since after the execution has started you can change the variable without any effect because the connections and signatures have been established when the blocks are inittialy passed to the scheduler (otherweise you would change the signature at runtime!).

So bottom line I think putting self.foo before or after the init of the basic_block is irrelevant, and I'm not sure what exactly does it solve.

tclarke, 2021-04-10:
Placing it before vs. after definitely has a different effect. I have not traced the entire execution path but the example specified behaves differently in grc in master when it's before vs. after.

dave-pi, 2021-04-10:
I'm trying to replicate it and it doesn't make any difference. Notice you are passing `vector_size` (and not `self.vector_size`) to `in_sig`, so `self.vector_size` being before, after, or even existing shouldn't have any effect.
Can you attach a flowgraph, the correspoding .py, and the epy_block_0.py?

skoslowski, 2021-04-10:
Duplicate of https://github.com/gnuradio/gnuradio/issues/4196

mormj, 2021-10-04:
Closing as duplicate.

closed 2021-10-04 by mormj