gnuradio/gnuradio #4510: filter.WIN_xxx changed to win_type.WIN_xxx

ChristopheSeguinot, 2021-04-09:
Hi 
I moved today from GR 3.8 to 3.9 after having to reinstall my linux computer. 

When I open this 3.8 grc file, [filter_taps.grc.txt](https://github.com/gnuradio/gnuradio/files/6286030/filter_taps.grc.txt),  under gnuradio 3.9 I get error related to filter windows. This window parameter look like `firdes.WIN_xxx` in 3.8, and should be renamed to  `window.WIN_xxx` in 3.9. 

In this example, I had to redefine each filter window. to avoid the error: 
```
Value "firdes.high_pass(gain, samp_rate, cutoff_freq, width, win, beta)" cannot be evaluated:
high_pass(): incompatible function arguments. The following argument types are supported:
    1. (gain: float, sampling_freq: float, cutoff_freq: float, transition_width: float, window: gnuradio.fft.fft_python.window.win_type = win_type.WIN_HAMMING, beta: float = 6.76) -> List[float]

Invoked with: 1.0, 32000, 2000, 1000, None, 6.76
```

This is quite annoying since I was expecting that 3.8 and 3.9 .grc files be compatible. 
Is there any idea to avoid such situations so that every flowgraph (created under 3.8 /3.9) could be opened with 3.9 or 3.8 ? 

Regards, Christophe

willcode, 2021-04-10:
The firdes names were intended to be removed in 3.8 but it did not happen until 3.9. Yes, an auto-converter could have been added for import into 3.9, but such things don't always get done. In any case, a search and replace in a text editor, changing `firdes.` to `window.` in the grc file is the easiest fix.

mbr0wn, 2021-05-14:
We don't really know how to fix this. Discussion welcome!

ChristopheSeguinot, 2021-05-15:
A temporary work around would be to adjust the loaded .grc file: 

- in GR 3.8 change every string `window.WIN_*` to` firdes.WIN_*`
- in GR 3.9 do the reverse change
This should allow user to use both 3.8 and 3.9 and revert to any of these versions and won't affect any other parameter but : is this a good practice ? 
However, this doesn't solve the issue when one previously generated .py file is directly launched by Python.

marcusmueller, 2024-01-28:
Impossible to fix, this was an intentional move.

closed 2024-01-28 by marcusmueller