gnuradio/gnuradio #4514: Source IO / Sink IO Compatibility

cclamb, 2021-04-09:
Hey everybody, it seems like the commit 7628a8f5a351faff88baa6de88f634b10aff3dfd in maint-38 is causing some issues. Specifically, whenever I try to attach a complex source to a sink on a USRP block, I end up getting an incompatible port error. Specifically, 'Source IO type "fc32" does not match sink IO type "complex"'. I've attached a flowgraph to demonstrate this - this installation is a fresh pybombs install from maint-38 with UHD-40 (though this failure also occurs at UHD-3.15 LTS).
[test_grc.zip](https://github.com/gnuradio/gnuradio/files/6288802/test_grc.zip)

willcode, 2021-04-09:
@mormj @mbr0wn @dl1ksv GRC does not know that "alias" types are the same as "core" types, so "fc32" is not the same as 'complex". Should we try to fix this or revert in all branches?

willcode, 2021-04-09:
Did not actually try it, but something like this could work:
```
diff --git a/grc/core/Connection.py b/grc/core/Connection.py
index 6fd78a41f..ddf402613 100644
--- a/grc/core/Connection.py
+++ b/grc/core/Connection.py
@@ -8,6 +8,7 @@ SPDX-License-Identifier: GPL-2.0-or-later
 
 
 from .base import Element
+from .Constants import ALIAS_OF
 from .utils.descriptors import lazy_property
 
 
@@ -92,7 +93,7 @@ class Connection(Element):
 
         source_dtype = self.source_port.dtype
         sink_dtype = self.sink_port.dtype
-        if source_dtype != sink_dtype:
+        if source_dtype != sink_dtype and source_dtype != ALIAS_OF(sink_dtype):
             self.add_error_message('Source IO type "{}" does not match sink IO type "{}".'.format(source_dtype, sink_dtype))
 
         source_size = self.source_port.item_size
diff --git a/grc/core/Constants.py b/grc/core/Constants.py
index b5b7c270c..65b4ac2ab 100644
--- a/grc/core/Constants.py
+++ b/grc/core/Constants.py
@@ -115,5 +115,21 @@ ALIAS_TYPES = {
     'bits':    (1, GRC_COLOR_PURPLE_A100),
 }
 
+ALIAS_OF = {
+    'complex': 'fc32',
+    'float': 'f32',
+    'int': 's32',
+    'short': 's16',
+    'byte': 's8',
+    'bits': 'bit',
+
+    'fc32': 'complex',
+    'f32': 'float',
+    's32': 'int',
+    's16': 'short',
+    's8': 'byte',
+    'bit': 'bits',
+}
+
 TYPE_TO_SIZEOF = {key: sizeof for name, key, sizeof, color in CORE_TYPES}
 TYPE_TO_SIZEOF.update((key, sizeof) for key, (sizeof, _) in ALIAS_TYPES.items())
```

dl1ksv, 2021-04-10:
@willcode I think we should fix this. I looked through the grc sources and found this:https://github.com/gnuradio/gnuradio/blob/d44ee73deeaab86a7cc5820ef62688f291331b88/grc/gui/Application.py#L352-L360

So I suppose we could handle this in a similar way. 
I'll try to install uhd and test the flowgraph.

willcode, 2021-04-10:
That only handles one alias in one direction. Application.py could use fixing later. Unfortunately, we don't know what aliases are in use in OOTs, so I think we have to allow all proper aliases.

closed 2021-04-10 by mormj

willcode, 2021-04-10:
@cclamb Thanks for catching this. Let us know if there is still a problem.

cclamb, 2021-04-10:
Looks good! Thanks everybody!

PR #4517 (2021-04-10): grc: match data type aliases as well as types (backport to maint-3.8)
A recent fix to validate connections by type, rather than by sizeof(type), exposed the fact that GRC did not
handle type aliases.

Fixes #4514

Backport https://github.com/gnuradio/gnuradio/pull/4516
PR #4518 (2021-04-10): grc: match data type aliases as well as types (backport to maint-3.9)
A recent fix to validate connections by type, rather than by sizeof(type), exposed the fact that GRC did not
handle type aliases.

Fixes #4514

Backport #4516