gnuradio/gnuradio #4519: Unused variable not defined in flowgraph cause GNURadio crashing

ChristopheSeguinot, 2021-04-10:
Hi, 
I unintentionally created a flowgraph ([test_variable_not_defined.grc.txt](https://github.com/gnuradio/gnuradio/files/6289949/test_variable_not_defined.grc.txt) ) in which variable `test1` was defined as `2*ovesampling1`, but `oversampling1` was undefined (after I deleted `ovesampling1` variable block )

- In the flowgraph, `test1` block is not red, so no one can't suspect the error. In fact, as `test1` is not used, this can be considered as a correct flowgraph. 
- Generating the flowgraph succeed. 
- while running the flowgraph crash GNURadio because of line `self.test1 = test1 = 2*oversampling1`

So one may conclude that: 

- whether this flowgraph should be considered as erroneous, because it contains undefined variable , and `test1` block should be highlighted in red, and python code not generated
- whether we consider this flowgraph as correct (I think this solution is more satisfying), in that case unused variable(s) should not appears when rendering the template

Sorry for rising an issue coming from this "stupid" flowgraph, but users could as me obtain such flowgraph by partially deleting unused blocks coming from a larger flowgraph. 

Regards, Christophe 

P.S. I will try to investigate this and propose a PR
[image]

dl1ksv, 2021-04-12:
I just tested your example and loading the flowgraph leads to 
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib64/python3.9/site-packages/gnuradio/grc/core/params/param.py", line 213, in evaluate
    value = self.parent_flowgraph.evaluate(expr)
  File "/usr/local/gnuradio/lib64/python3.9/site-packages/gnuradio/grc/core/FlowGraph.py", line 292, in evaluate
    return self._eval_cache.setdefault(expr, eval(expr, self.namespace, local_namespace))
  File "<string>", line 1, in <module>
NameError: name 'oversampling1' is not defined

And the same happens if I try to run the flowgraph but grc does not crash. This error should be catched and make the fg invalid.

ChristopheSeguinot, 2021-04-12:
About the crash, it's just a crash report emitted (? by python) indicating errors, but effectively GnuradioCompanion keep running.

Yes, after my investigation this morning, I think it' better to catch this error and mark the fg as invalid. 

I made different test this morning, and coming back to the faulty fg, I now surprisingly observe that the faulty block turned red (this was not the case before), but the fg is not marked as invalid (if a block is invalid, shouldn't the fg be marked as invalid ?).

dl1ksv, 2021-04-12:
As the variable test1 cannot be evaluteted, I think and this block should go red and the button 'View flow graph errors' should be enabled.

ChristopheSeguinot, 2021-04-12:
I don't find or understand exactly where the block is marked invalid. 
Each block is validated in     def validate(self): https://github.com/gnuradio/gnuradio/blob/5547665ee92f748a7ee47d64dfbf133db77fcfce/grc/core/blocks/block.py#L228
which call 
https://github.com/gnuradio/gnuradio/blob/5547665ee92f748a7ee47d64dfbf133db77fcfce/grc/core/blocks/block.py#L275
which call 
https://github.com/gnuradio/gnuradio/blob/5547665ee92f748a7ee47d64dfbf133db77fcfce/grc/core/FlowGraph.py#L269

```
    def evaluate(self, expr, namespace=None, local_namespace=None):
        """
        Evaluate the expression.
        """
        # Evaluate
        if not expr:
            raise Exception('Cannot evaluate empty statement.')
        if namespace is not None:
            return eval(expr, namespace, local_namespace)
        else:
            return self._eval_cache.setdefault(expr, eval(expr, self.namespace, local_namespace))
```

When loading my faulty flowgraph, considering variable test1 whose value is `2*oversampling` 
`evaluate(..)` and  `return self._eval_cache.setdefault(expr, eval(expr, self.namespace, local_namespace))`

- are called 5 times before the end of fg loading (message >>>Done), whitout any error or exception raised
- are called 7 times after message >>>Done, 
- and finally the error occurs: 
```
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/FlowGraph.py", line 283, in evaluate
    return self._eval_cache.setdefault(expr, eval(expr, self.namespace, local_namespace)) 
  File "<string>", line 1, in <module>
NameError: name 'oversampling' is not defined

```
I don't understand why the error or exception is not raised before !

dl1ksv, 2021-04-13:
The error should appear here
https://github.com/gnuradio/gnuradio/blob/5547665ee92f748a7ee47d64dfbf133db77fcfce/grc/core/params/param.py#L137-L141 and make the flowgraph invalid

ChristopheSeguinot, 2021-04-13:
for faulty blocks: 

- The self.add_error_message(str(e)) is fired 
- using self.get_error_messages() I verified that the error is "registered"
- using self.is_valid() I verified the block are marked as invalid

However, for variable blocks: 
- the error is not listed in the GNURadioCompanion "errors and warnings list"
- the faulty variable block doesn't turn red and don't have any error message displayed

I'm steel searching how the block are redrawn when an error occurs.

dl1ksv, 2021-04-13:
Meanwhile I found where the unhandled exception occurs, It is
https://github.com/gnuradio/gnuradio/blob/5547665ee92f748a7ee47d64dfbf133db77fcfce/grc/gui/VariableEditor.py#L181-L189

Modifying to

                    for key in block.params:
                        try:
                            evaluated = str(block.params[key].evaluate())
                            # ensure that evaluated is a UTF-8 string
                            # Old PMTs could produce non-UTF-8 strings
                            evaluated = evaluated.encode('utf-8', 'backslashreplace').decode('utf-8')
                            self.set_tooltip_text(evaluated)
                        except Exception as error:
                            #import pdb;pdb.set_trace()
                            block.add_error_message(str(error))
                            block.params[key].add_error_message(str(error))
                            self.set_tooltip_text(str(error))
                            pass

leads to
[image]
[image]

But the variable itself does not turn to red.  If you start grc the ' display error'  button is disabled.   But if you click into the flowgraph the button is enabled.

ChristopheSeguinot, 2021-04-14:
Thanks Volker, we are no closer to the solution. 

I should says that I looked deeply into the GR code, without understanding the way flowgraph are validated. The fact that each variable is evaluated 2 or 4 time is mysterous. And the fact that this evaluation are redone every time the flowgraph is move, or the window resized seem unusefull.   

From what I learned: 
- It looks like "variable blocks" behave differently from other blocks (which derived from GTK objects). 
- I didn't found any way to make erroneous variables block label to turn red. Look like this is not implemented (no method) for these blocks! 
- I even didn't find a proper way to directly alter the block Variable label.

ChristopheSeguinot, 2021-05-17:
Volker PR #4629 solves the issue. Thanks

PR #4629 (2021-06-01): grc: add validation for variables of type raw
At the moment grc does not validate variables, as they are of type raw.
So they are always treated as valid, which leads to some problems, see: #4519.

A approach to fix #4519 is #4531.
This approach improves the situation, but has still some problems.

param.py provides a validator function for dtype raw, but this is not used, as dtypes.py provides no validator function for dtype raw.

This fix introduces a validator function in dtypes.py which calls the corresponding function in param.py.

In addition this makes it possible, to make the code in V [... cut]
commit e562d6b3 (2021-06-01): grc: handle variables of type raw correctly

At the moment grc does not validate variables, as they are of type raw.
So they are always treated as valid, which leads to some problems, see: #4519.

A approach to fix #4519 is #4531.
This approach improves the situation, but has still some problems.

param.py provides a validator function for dtype raw, but this is not used, as dtypes.py provides no validator function for dtype raw.

This fix clears the error messages in base.rewrite not in base.elements.validate, so the error messages are kept.

In addition this makes it possible, to make the co [... cut]
PR #4697 (2021-06-01): grc: handle variables of type raw correctly (backport to maint-3.9)
At the moment grc does not validate variables, as they are of type raw.
So they are always treated as valid, which leads to some problems, see: #4519.

A approach to fix #4519 is #4531.
This approach improves the situation, but has still some problems.

param.py provides a validator function for dtype raw, but this is not used, as dtypes.py provides no validator function for dtype raw.

This fix clears the error messages in base.rewrite not in base.elements.validate, so the error messages are kept.

In addition this makes it p [... cut]
commit 1967f5fa (2021-06-01): grc: handle variables of type raw correctly

At the moment grc does not validate variables, as they are of type raw.
So they are always treated as valid, which leads to some problems, see: #4519.

A approach to fix #4519 is #4531.
This approach improves the situation, but has still some problems.

param.py provides a validator function for dtype raw, but this is not used, as dtypes.py provides no validator function for dtype raw.

This fix clears the error messages in base.rewrite not in base.elements.validate, so the error messages are kept.

In addition this makes it possible, to make the co [... cut]
closed 2021-05-17 by ChristopheSeguinot