gnuradio/gnuradio #4524: ImportError: cannot import name 'square_ff' from 'howto' (unknown location)

doNotBeTooSerious, 2021-04-12:
Hi, there, I've been trying to create my first OOT named 'howto' according to the guidelines given in [.](https://wiki.gnuradio.org/index.php/OutOfTreeModules) The first steps are fine until the 'make test' command.
I run the 'ctest -V -R' to check detailed info, it says 'howto' module is not found.
I then use the 'export PYTHONPATH=$PYTHONPATH:/usr/local/lib/cmake/' command to add the 'howto' folder path to python path, but the 'ctest -V -R' then yeilds 'ImportError: cannot import name 'square_ff' from 'howto' (unknown location)'.
As I said, the previous 'cmake ../' and 'make' went well and I am sure I have run the 'sudo make install' command which creats the 'howto' folder under path '/usr/local/lib/cmake/', I believe. My 'howto' module folder (created by 'gr_modtool newmod howto' command, under the /home/tem/ path).
So I am not where I've done wrong, any suggestions or similar problems?
Many `thanks!
Let me know if you wish for any further information.
Below is my CMakeCache.TXT file: 
`# This is the CMakeCache file.
# For build in directory: /home/michael/tem/gr-howto/build
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

Boost_DATE_TIME_LIBRARY_RELEASE:STRING=/usr/lib/x86_64-linux-gnu/libboost_date_time.so.1.71.0

//The directory containing a CMake configuration file for Boost.
Boost_DIR:PATH=/usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0

//Path to a file.
Boost_INCLUDE_DIR:PATH=/usr/include

Boost_PROGRAM_OPTIONS_LIBRARY_RELEASE:STRING=/usr/lib/x86_64-linux-gnu/libboost_program_options.so.1.71.0

Boost_REGEX_LIBRARY_RELEASE:STRING=/usr/lib/x86_64-linux-gnu/libboost_regex.so.1.71.0

Boost_SYSTEM_LIBRARY_RELEASE:STRING=/usr/lib/x86_64-linux-gnu/libboost_system.so.1.71.0

Boost_THREAD_LIBRARY_RELEASE:STRING=/usr/lib/x86_64-linux-gnu/libboost_thread.so.1.71.0

Boost_UNIT_TEST_FRAMEWORK_LIBRARY_RELEASE:STRING=/usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.71.0

//Path to a program.
CMAKE_ADDR2LINE:FILEPATH=/usr/bin/addr2line

//Path to a program.
CMAKE_AR:FILEPATH=/usr/bin/ar

//Choose the type of build, options are: None Debug Release RelWithDebInfo
// MinSizeRel ...
CMAKE_BUILD_TYPE:STRING=

//Enable/Disable color output during build.
CMAKE_COLOR_MAKEFILE:BOOL=ON

//CXX compiler
CMAKE_CXX_COMPILER:FILEPATH=/usr/bin/c++

//A wrapper around 'ar' adding the appropriate '--plugin' option
// for the GCC compiler
CMAKE_CXX_COMPILER_AR:FILEPATH=/usr/bin/gcc-ar-9

//A wrapper around 'ranlib' adding the appropriate '--plugin' option
// for the GCC compiler
CMAKE_CXX_COMPILER_RANLIB:FILEPATH=/usr/bin/gcc-ranlib-9

//Flags used by the CXX compiler during all build types.
CMAKE_CXX_FLAGS:STRING=

//Flags used by the CXX compiler during DEBUG builds.
[... 542 lines cut ...]
_Python:INTERNAL=PYTHON
__pkg_config_arguments_PC_GMP:INTERNAL=gmp
__pkg_config_checked_PC_GMP:INTERNAL=1
//ADVANCED property for variable: boost_atomic_DIR
boost_atomic_DIR-ADVANCED:INTERNAL=1
//ADVANCED property for variable: boost_date_time_DIR
boost_date_time_DIR-ADVANCED:INTERNAL=1
//ADVANCED property for variable: boost_headers_DIR
boost_headers_DIR-ADVANCED:INTERNAL=1
//ADVANCED property for variable: boost_program_options_DIR
boost_program_options_DIR-ADVANCED:INTERNAL=1
//ADVANCED property for variable: boost_regex_DIR
boost_regex_DIR-ADVANCED:INTERNAL=1
//ADVANCED property for variable: boost_system_DIR
boost_system_DIR-ADVANCED:INTERNAL=1
//ADVANCED property for variable: boost_thread_DIR
boost_thread_DIR-ADVANCED:INTERNAL=1
//ADVANCED property for variable: boost_unit_test_framework_DIR
boost_unit_test_framework_DIR-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_PC_GMP_gmp
pkgcfg_lib_PC_GMP_gmp-ADVANCED:INTERNAL=1
prefix_result:INTERNAL=/usr/lib/x86_64-linux-gnu
//Directories where pybind11 and possibly Python headers are located
pybind11_INCLUDE_DIRS:INTERNAL=/usr/local/include;/usr/include/python3.8
`

doNotBeTooSerious, 2021-04-13:
Sorry for uploading this issue here, I add the python path, and now it works. However, when I add the block to GNU radio, it appears that I can see the module but it would fail if I drag it out, does it also have something to do with the OOT module path? do I need to run gr_modtool under path /use/local/ or I need to reinstall gnuradio to a specific path to avoid such problems?

There are where I find my howto packages now after I run the find /home/michael/prefix-3.8 -name howto :
home/michael/prefix-3.8/include/howto
/home/michael/prefix-3.8/lib/python3/dist-packages/howto
/home/michael/prefix-3.8/lib/python3.8/site-packages/howto
/home/michael/prefix-3.8/lib/cmake/howto

velichkov, 2021-04-13:
> Sorry for uploading this issue here, 

No problem, but note that you will get help and answers a lot faster in chat.gnuradio.org or in the mailing list. The github issues are more suitable for bug reports.

> I add the python path, and now it works. 

That's great! 

> However, when I add the block to GNU radio, it appears that I can see the module but it would fail if I drag it out, does it also have something to do with the OOT module path? 

When you start the gnuradio companion it prints the block paths in its console (the bottom left corner). What paths do you see there? When you drag your OOT block do you see any errors in the console. 

Also execute the `gnuradio-companion` in the terminal, drag your block and check for errors in the terminal.

> do I need to run gr_modtool under path /use/local/ 

No. You may try installing your OOT module in `/usr/local/` but for this you don't need to run gr_modtool. You need to go in the build directory of your OOT module, delete the CMakeCache.txt file and then re-run `cmake` without specifying the install prefix (the default is `/usr/local`). I don't know if this is going to solve your problem or not. 

> I need to reinstall gnuradio to a specific path to avoid such problems?

No. How did you install gnuradio, which version and Linux distribution?

> Thanks for your answer and maybe I will refer further problems to the mailing list or IRC/Matrix chat.

My advice is to close this issue as this is not gnuradio's bug and post your questions to IRC/Matrix.

> There are where I find my howto packages now after I run the find /home/michael/prefix-3.8 -name howto :
> home/michael/prefix-3.8/include/howto
> /home/michael/prefix-3.8/lib/python3/dist-packages/howto
> /home/michael/prefix-3.8/lib/python3.8/site-packages/howto
> /home/michael/prefix-3.8/lib/cmake/howto

It's a bit disturbing that you have your OOT module in both `dist-packages` and `site-packages`. Which Linux distribution do you use? If I were you, I would delete the whole `/home/michael/prefix-3.8` directory and then reinstall the module in a clean prefix. One advantage in having your OOT module installed in a custom prefix (not in `/usr/local/`) is that you can just wipe out the whole directory and start from scratch.

closed 2021-04-13 by doNotBeTooSerious