gnuradio/gnuradio #4525: Create GitHub issue templates

ei8fdb, 2021-04-12:
(work-in-progress ticket)

We should have GH issue templates for:

- reports for bugs
- reports for usability improvements
- requests for new features

ryanvolz, 2021-05-14:
Adding from hackfest discussion: an issue template would be a good place to remind people that requests for user support don't belong in an issue, and instead they should go to the Github Discussions / mailing list / matrix room.

mbr0wn, 2021-05-14:
@devnulling wrote this on the original issue:

> 
> The awesome folks over at Ettus Research have a fancy Issue and PR template. We should use one like it.
> 
> https://raw.githubusercontent.com/EttusResearch/uhd/master/.github/ISSUE_TEMPLATE.md
> 
> https://raw.githubusercontent.com/EttusResearch/uhd/master/.github/pull_request_template.md

ei8fdb, 2021-05-15:
I have an issue template ready in a repo (I just needed to figure out how to send the PR) but I'll rewrite it to incorporate the useful bits from ettus template.

Their PR template makes sense too.

mormj, 2022-01-20:
@ei8fdb - thanks for spearheading this - looks like your template recommendations got merged in some form, so closing this issue.  Of course if you see improvements that could be made, please submit PRs with fixes