gnuradio/gnuradio #4527: GNU Radio on ARM

rodrirq, 2021-04-13:
Hello,

I am trying to crosscompile GNU Radio from Linux to ARM (Xilinx Zynq Z7-10). I am having problems with the boost library. I tried cross-compiling it too but, after the cmake, it says that it cannot find the library.

I hope someone knows what to do, thank you.

balister, 2021-04-14:
I do this a lot, what is your exact issue. I had some old notes on my website, but it is broken at the moment, I am trying to fix it.

rodrirq, 2021-04-15:
Hello @balister, 

I do:

$ git clone --recursive https://github.com/gnuradio/gnuradio.git

$ cd gnuradio

$ git checkout maint-3.7

$ git pull --recurse-submodules=on

$ git submodule update --init

$ cp volk/cmake/Toolchains/arm-linux-gnueabihf.cmake cmake/Toolchains/

I have problems with the boost dependencies. I downloaded boost from: [https://www.boost.org/users/history/version_1_75_0.html](url). 

$ cd Downloads

$ tar xzvf boost_1_75_0.tar.gz boost_1_75_0

$ cd boost_1_75_0

$ BOOST_PREFIX=/opt/boost_1_75_0 

$ ./bootstrap.sh

$ sudo ./b2 --prefix=$BOOST_PREFIX --with-thread --with-date_time --with-program_options --with-filesystem --with-system --layout=versioned threading=multi variant=release install 

This last step appears on the README.building-boost

Then I do:

$ cd gnuradio

$ mkdir build

$ cd build

$ cmake -DBOOST_ROOT=$BOOST_PREFIX -DBoost_INCLUDE_DIR=$BOOST_PREFIX/include/boost-1_75/ -DBoost_LIBRARY_DIRS=$BOOST_PREFIX/lib -DCMAKE_TOOLCHAIN_FILE=../cmake/Toolchains/arm-linux-gnueabihf.cmake ../

This last step appears on the same README file. Wit this, I am able to cross compile it.

In the end, I get this result:

-- ######################################################
-- # Gnuradio enabled components                         
-- ######################################################
--   * volk
--   * doxygen
--   * sphinx
--   * gnuradio-runtime
--   * gr-ctrlport
--   * gr-blocks
--   * gr-fec
--   * gr-fft
--   * gr-filter
--   * gr-analog
--   * gr-digital
--   * gr-dtv
--   * gr-atsc
--   * gr-audio
--   * * oss
--   * gr-comedi
--   * gr-channels
--   * gr-noaa
--   * gr-pager
--   * gr-trellis
--   * gr-vocoder
--   * gr-wavelet
--   * gr-zeromq
-- 
-- ######################################################
-- # Gnuradio disabled components                        
-- ######################################################
--   * python-support
--   * testing-support
--   * gnuradio-companion
--   * gr-qtgui
--   * gr-uhd
--   * gr-utils
--   * gr-video-sdl
--   * gr-fcd
--   * gr-wxgui

The thing is that, when i do $ make, I get this error: 

[  5%] Linking CXX executable volk_profile
CMakeFiles/volk_profile.dir/volk_profile.cc.o: In function `write_results(std::vector<volk_test_results_t, std::allocator<volk_test_results_t> > const*, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
volk_profile.cc:(.text+0x51ba): undefined reference to `boost::filesystem::path::parent_path() const'
volk_profile.cc:(.text+0x51c6): undefined reference to `boost::filesystem::detail::status(boost::filesystem::path const&, boost::system::error_code*)'
volk_profile.cc:(.text+0x51fa): undefined reference to `boost::filesystem::path::parent_path() const'
volk_profile.cc:(.text+0x5274): undefined reference to `boost::filesystem::path::parent_path() const'
volk_profile.cc:(.text+0x527c): undefined reference to `boost::filesystem::detail::create_directories(boost::filesystem::path const&, boost::system::error_code*)'
collect2: error: ld returned 1 exit status
volk/apps/CMakeFiles/volk_profile.dir/build.make:147: recipe for target 'volk/apps/volk_profile' failed
make[2]: *** [volk/apps/volk_profile] Error 1
CMakeFiles/Makefile2:346: recipe for target 'volk/apps/CMakeFiles/volk_profile.dir/all' failed
make[1]: *** [volk/apps/CMakeFiles/volk_profile.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

I hope you know the way to resolve this. Thank you in advance.

Rodrigo.

balister, 2021-04-15:
What toolchain are you using?

rodrirq, 2021-04-15:
@balister I am using this Toolchain.

It doesn't appear in the gnuradio's toolchain folder. But it does appear in the volk folder.

I copied it with: $ cp volk/cmake/Toolchains/arm-linux-gnueabihf.cmake cmake/Toolchains/

**arm-linux-gnueabihf.cmake**

set(CMAKE_SYSTEM_NAME Linux)
set(CMAKE_SYSTEM_PROCESSOR ARM)

if(MINGW OR CYGWIN OR WIN32)
  set(UTIL_SEARCH_CMD where)
elseif(UNIX OR APPLE)
  set(UTIL_SEARCH_CMD which)
endif()

set(TOOLCHAIN_PREFIX arm-linux-gnueabihf-)

execute_process(
  COMMAND ${UTIL_SEARCH_CMD} ${TOOLCHAIN_PREFIX}gcc
  OUTPUT_VARIABLE BINUTILS_PATH
  OUTPUT_STRIP_TRAILING_WHITESPACE
  )

get_filename_component(ARM_TOOLCHAIN_DIR ${BINUTILS_PATH} DIRECTORY)

# The following is not needed on debian
# Without that flag CMake is not able to pass test compilation check
#set(CMAKE_EXE_LINKER_FLAGS_INIT "--specs=nosys.specs")

set(CMAKE_C_COMPILER ${TOOLCHAIN_PREFIX}gcc)
set(CMAKE_ASM_COMPILER ${CMAKE_C_COMPILER})
set(CMAKE_CXX_COMPILER ${TOOLCHAIN_PREFIX}g++)

set(CMAKE_OBJCOPY ${ARM_TOOLCHAIN_DIR}/${TOOLCHAIN_PREFIX}objcopy CACHE INTERNAL "objcopy tool")
set(CMAKE_SIZE_UTIL ${ARM_TOOLCHAIN_DIR}/${TOOLCHAIN_PREFIX}size CACHE INTERNAL "size tool")

set(CMAKE_FIND_ROOT_PATH ${BINUTILS_PATH})

set(CMAKE_CROSSCOMPILING_EMULATOR "qemu-arm -L /usr/arm-linux-gnueabihf/")

rodrirq, 2021-04-16:
Hello @balister,

I tried to change my toolchain. I used the oe-sdk_cross.cmake

````
set( CMAKE_SYSTEM_NAME Linux )
set( CMAKE_C_COMPILER /usr/bin/arm-linux-gnueabihf-gcc )
set( CMAKE_CXX_COMPILER /usr/bin/arm-linux-gnueabihf-g++ )
#string(REGEX MATCH "sysroots/([a-zA-Z0-9]+)" CMAKE_SYSTEM_PROCESSOR $ENV{SDKTARGETSYSROOT})
#string(REGEX REPLACE "sysroots/" "" CMAKE_SYSTEM_PROCESSOR ${CMAKE_SYSTEM_PROCESSOR})
set( CMAKE_CXX_FLAGS $ENV{CXXFLAGS}  CACHE STRING "" FORCE )
set( CMAKE_C_FLAGS $ENV{CFLAGS} CACHE STRING "" FORCE ) #same flags for C sources
set( CMAKE_LDFLAGS_FLAGS ${CMAKE_CXX_FLAGS} CACHE STRING "" FORCE ) #same flags for C sources
set( CMAKE_LIBRARY_PATH ${OECORE_TARGET_SYSROOT}/usr/lib )
set( CMAKE_FIND_ROOT_PATH $ENV{OECORE_TARGET_SYSROOT} $ENV{OECORE_NATIVE_SYSROOT} )
set( CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER )
set( CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY )
set( CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY )
set ( ORC_INCLUDE_DIRS $ENV{OECORE_TARGET_SYSROOT}/usr/include/orc-0.4 )
set ( ORC_LIBRARY_DIRS $ENV{OECORE_TARGET_SYSROOT}/usr/lib )
````

I commented the strings, because they failed. I get the same result.

Then I saw https://github.com/gnuradio/volk/pull/235 and modified the volk_profile.cc as explained.

Now I don't get the boost error, but I get the following one:

````
[  8%] Linking CXX executable volk_test_all
[  8%] Linking CXX executable volk_profile
[  8%] Linking CXX executable volk-config-info
[  8%] Linking CXX shared library libgnuradio-pmt-3.7.14.0.so
CMakeFiles/volk_profile.dir/volk_profile.cc.o: In function `write_results(std::vector<volk_test_results_t, std::allocator<volk_test_results_t> > const*, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
volk_profile.cc:(.text+0xb96): undefined reference to `std::experimental::filesystem::v1::__cxx11::path::_M_split_cmpts()'
volk_profile.cc:(.text+0xba0): undefined reference to `std::experimental::filesystem::v1::__cxx11::path::parent_path() const'
volk_profile.cc:(.text+0xbaa): undefined reference to `std::experimental::filesystem::v1::status(std::experimental::filesystem::v1::__cxx11::path const&)'
volk_profile.cc:(.text+0xc76): undefined reference to `std::experimental::filesystem::v1::__cxx11::path::parent_path() const'
volk_profile.cc:(.text+0xd32): undefined reference to `std::experimental::filesystem::v1::__cxx11::path::parent_path() const'
volk_profile.cc:(.text+0xd38): undefined reference to `std::experimental::filesystem::v1::create_directories(std::experimental::filesystem::v1::__cxx11::path const&)'
collect2: error: ld returned 1 exit status
[  8%] Built target volk-config-info
[  8%] Built target volk_test_all
volk/apps/CMakeFiles/volk_profile.dir/build.make:147: recipe for target 'volk/apps/volk_profile' failed
make[2]: *** [volk/apps/volk_profile] Error 1
CMakeFiles/Makefile2:346: recipe for target 'volk/apps/CMakeFiles/volk_profile.dir/all' failed
make[1]: *** [volk/apps/CMakeFiles/volk_profile.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  8%] Built target gnuradio-pmt
/home/rodrirq/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_sink.h:129: warning: explicit link request to 'uhd::device_addr_t' could not be resolved
/home/rodrirq/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_sink.h:129: warning: explicit link request to 'uhd::device_addr_t' could not be resolved
/home/rodrirq/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_source.h:88: warning: explicit link request to 'uhd::device_addr_t' could not be resolved
/home/rodrirq/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_source.h:88: warning: explicit link request to 'uhd::device_addr_t' could not be resolved
/home/rodrirq/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_sink.h:129: warning: explicit link request to 'uhd::device_addr_t' could not be resolved
/home/rodrirq/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_source.h:88: warning: explicit link request to 'uhd::device_addr_t' could not be resolved
/home/rodrirq/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_sink.h:129: warning: explicit link request to 'uhd::device_addr_t' could not be resolved
/home/rodrirq/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_source.h:88: warning: explicit link request to 'uhd::device_addr_t' could not be resolved
sh: latex: command not found
error: Problems running latex. Check your installation or look for typos in _formulas.tex and check _formulas.log!
sh: dvips: command not found
error: Problems running dvips. Check your installation!
/home/rodrirq/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_sink.h:129: warning: explicit link request to 'uhd::device_addr_t' could not be resolved
/home/rodrirq/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_source.h:88: warning: explicit link request to 'uhd::device_addr_t' could not be resolved
/home/rodrirq/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_sink.h:129: warning: explicit link request to 'uhd::device_addr_t' could not be resolved
/home/rodrirq/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_source.h:88: warning: explicit link request to 'uhd::device_addr_t' could not be resolved
[  8%] Built target doxygen_target
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
````

balister, 2021-04-20:
What Linux distribution are you using on your zynq? What compiler are you using?

rodrirq, 2021-04-20:
@balister I have Ubuntu 18.04.5 on my computer, and my Zynq has PetaLinux 2020.2.
 
I am compiling with arm-linux-gnueabihf-gcc and arm-linux-gnueabihf-g++.

I created a new post https://github.com/gnuradio/gnuradio/issues/4546 where I asked if using an external volk is a possibility, because the problem is with the volk_profile.cc.

balister, 2021-04-20:
Where did those compilers come from?

rodrirq, 2021-04-20:
@balister I don't understand your question. 

I used: `$ sudo apt-get install gcc-arm-linux-gnueabihf`. Same for g++.

I don't know if that's the answer you were expecting.

rodrirq, 2021-04-20:
@balister I have changed my compilers to gcc and g++.

It compiled, now I am running the make and it seems I don't get the volk_profile.cc error. 

It is running right now, I will keep updating after it finishes.

balister, 2021-04-20:
You are using the ubuntu crosscompilers, you need to use the sdk from petalinux. The compiler from Ubuntu will not work.

rodrirq, 2021-04-20:
@balister I will need to investigate on that, because I don't have much idea about how to use the SDK and where it is.

Thank you for your answer.

rodrirq, 2021-04-21:
Hello @balister,

After doing petalinux-config -c kernel, I got to see the compiler:

[image]

It says that my compiler is arm-xilinx-linux-gnueabi-gcc. I don't know if this is what you meant, it is my first time doing this, so I am a bit lost. If it's that compiler, how do I use it? I have checked the folders from my petalinux project and I don't find it.

I somehow managed to cross-compile LimeSuite with the compilers arm-linux-gnueabihf-gcc and g++, so I thought those were the correct ones. 

Sorry for the inconveniences.

rodrirq, 2021-04-21:
@balister I have seen this post: https://github.com/phase4ground/DVB-receiver/wiki/Adding-gnuradio-to-the-Petalinux-2018.2-BSP-based-build-for-the-Ultra96-board

I don't know if it would work, but I can't do the git submodule add part, it says:
`fatal: not a git repository (or any of the parent directories): .git`

rodrirq, 2021-04-29:
Hello @balister, I have been studying how SDK works, and I have finally succesfully installed it. I have found the cross-compilers on /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/bin/arm-xilinx-linux-gnueabi.
I have cross-compiled an external Volk, but when I try to cmake the GNU Radio, I get this output:
```
$ cmake -Wno-dev -DCMAKE_TOOLCHAIN_FILE=/home/rodrirq/gnuradio/cmake/Toolchains/oe-sdk_cross.cmake -DENABLE_INTERNAL_VOLK=OFF -DENABLE_GR_WXGUI=OFF -DENABLE_GR_VOCODER=OFF -DENABLE_GR_DTV=OFF -DENABLE_GR_ATSC=OFF -DENABLE_DOXYGEN=OFF ../ 
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- Check for working C compiler: /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/bin/arm-xilinx-linux-gnueabi/arm-xilinx-linux-gnueabi-gcc
-- Check for working C compiler: /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/bin/arm-xilinx-linux-gnueabi/arm-xilinx-linux-gnueabi-gcc -- broken
CMake Error at /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/share/cmake-3.15/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/bin/arm-xilinx-linux-gnueabi/arm-xilinx-linux-gnueabi-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/rodrirq/volk/build/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/make cmTC_87b7d/fast && /usr/bin/make -f CMakeFiles/cmTC_87b7d.dir/build.make CMakeFiles/cmTC_87b7d.dir/build
    make[1]: Entering directory '/home/rodrirq/volk/build/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_87b7d.dir/testCCompiler.c.o
    /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/bin/arm-xilinx-linux-gnueabi/arm-xilinx-linux-gnueabi-gcc   -O2 -pipe -g -feliminate-unused-debug-types     -o CMakeFiles/cmTC_87b7d.dir/testCCompiler.c.o   -c /home/rodrirq/volk/build/CMakeFiles/CMakeTmp/testCCompiler.c
    Linking C executable cmTC_87b7d
    /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_87b7d.dir/link.txt --verbose=1
    /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/bin/arm-xilinx-linux-gnueabi/arm-xilinx-linux-gnueabi-gcc  -O2 -pipe -g -feliminate-unused-debug-types    -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed  CMakeFiles/cmTC_87b7d.dir/testCCompiler.c.o  -o cmTC_87b7d 
    /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/libexec/arm-xilinx-linux-gnueabi/gcc/arm-xilinx-linux-gnueabi/9.2.0/real-ld: cannot find Scrt1.o: No such file or directory
    /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/libexec/arm-xilinx-linux-gnueabi/gcc/arm-xilinx-linux-gnueabi/9.2.0/real-ld: cannot find crti.o: No such file or directory
    /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/libexec/arm-xilinx-linux-gnueabi/gcc/arm-xilinx-linux-gnueabi/9.2.0/real-ld: cannot find crtbeginS.o: No such file or directory
    /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/libexec/arm-xilinx-linux-gnueabi/gcc/arm-xilinx-linux-gnueabi/9.2.0/real-ld: cannot find -lgcc
    /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/libexec/arm-xilinx-linux-gnueabi/gcc/arm-xilinx-linux-gnueabi/9.2.0/real-ld: cannot find -lgcc_s
    /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/libexec/arm-xilinx-linux-gnueabi/gcc/arm-xilinx-linux-gnueabi/9.2.0/real-ld: cannot find -lc
    /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/libexec/arm-xilinx-linux-gnueabi/gcc/arm-xilinx-linux-gnueabi/9.2.0/real-ld: cannot find -lgcc
    /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/libexec/arm-xilinx-linux-gnueabi/gcc/arm-xilinx-linux-gnueabi/9.2.0/real-ld: cannot find -lgcc_s
    /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/libexec/arm-xilinx-linux-gnueabi/gcc/arm-xilinx-linux-gnueabi/9.2.0/real-ld: cannot find crtendS.o: No such file or directory
    /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/libexec/arm-xilinx-linux-gnueabi/gcc/arm-xilinx-linux-gnueabi/9.2.0/real-ld: cannot find crtn.o: No such file or directory
    collect2: error: ld returned 1 exit status
    CMakeFiles/cmTC_87b7d.dir/build.make:86: recipe for target 'cmTC_87b7d' failed
    make[1]: *** [cmTC_87b7d] Error 1
    make[1]: Leaving directory '/home/rodrirq/volk/build/CMakeFiles/CMakeTmp'
    Makefile:121: recipe for target 'cmTC_87b7d/fast' failed
    make: *** [cmTC_87b7d/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:23 (project)

-- Configuring incomplete, errors occurred!
See also "/home/rodrirq/volk/build/CMakeFiles/CMakeOutput.log".
See also "/home/rodrirq/volk/build/CMakeFiles/CMakeError.log".

```

This is my toolchain:
```
set( CMAKE_SYSTEM_NAME Linux )
set( CMAKE_C_COMPILER /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/bin/arm-xilinx-linux-gnueabi/arm-xilinx-linux-gnueabi-gcc )
set( CMAKE_CXX_COMPILER /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/bin/arm-xilinx-linux-gnueabi/arm-xilinx-linux-gnueabi-g++ )
#string(REGEX MATCH "sysroots/([a-zA-Z0-9]+)" CMAKE_SYSTEM_PROCESSOR $ENV{SDKTARGETSYSROOT})
#string(REGEX REPLACE "sysroots/" "" CMAKE_SYSTEM_PROCESSOR ${CMAKE_SYSTEM_PROCESSOR})
set( CMAKE_CXX_FLAGS $ENV{CXXFLAGS}  CACHE STRING "" FORCE )
set( CMAKE_C_FLAGS $ENV{CFLAGS} CACHE STRING "" FORCE ) #same flags for C sources
set( CMAKE_LDFLAGS_FLAGS ${CMAKE_CXX_FLAGS} CACHE STRING "" FORCE ) #same flags for C sources
set( CMAKE_LIBRARY_PATH ${OECORE_TARGET_SYSROOT}/usr/lib )
[... cut ...]
set( CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY )
set( ORC_INCLUDE_DIRS $ENV{OECORE_TARGET_SYSROOT}/usr/include/orc-0.4 )
set( ORC_LIBRARY_DIRS $ENV{OECORE_TARGET_SYSROOT}/usr/lib )
set( BOOST_INCLUDEDIR /opt/boost_1_76_0/include/boost-1_76/boost )
set( BOOST_LIBRARYDIR /opt/boost_1_76_0/lib )
set(CMAKE_ASM_FLAGS "${CMAKE_CXX_FLAGS} -g" CACHE STRING "" FORCE) #same flags for asm sources 

```

I hope you can help me with this. I looked for the libraries but I have already installed them, so I don't know where is the error.

closed 2021-05-03 by rodrirq