gnuradio/gnuradio #4528: OFDM Receiver not logging debug info: data files cannit be exported

RosalindAgasti, 2021-04-13:
I was trying to execute the ofdm_tx flowgraph in GNURadio to obtain the channel estimation data from the OFDM receiver block but it fails to execute and there seems to be an error in the python file. I have attached the error message here.

Is there a way to fix this bug?

[image]

marcusmueller, 2021-04-18:
Hi @RosalindAgasti!

Great to have you here. I think there's multiple things to unpack here:

1. We might need to know which version of GNU Radio you have. I'll try with a few versions that I suspect are candidates, but knowing your definite version would be good.
2. The error should indeed be fixed; looks like it's in a hierarchical block that GNU Radio itself ships: that's bad!
3. The logging should be easy to add, but I think it'd be a separate feature request issue, if I understood you correctly.

mbr0wn, 2021-05-14:
@RosalindAgasti I tried running the ofdm_tx.grc, and it worked fine. I'm going to close this issue, but please reopen it if you still have problems. But if you do, add your GNU Radio version and what exactly you did to trigger the error.

closed 2021-05-14 by mbr0wn