gnuradio/gnuradio #4545: gr_waveform_t.GR_COS_WAVE is invalid for type 'int'

dfy888, 2021-04-20:
Error: gr_waveform_t.GR_COS_WAVE is invalid for type 'int'

install way :  sudo apt-get install gnuradio       version==3.9.0

dkozel, 2021-04-20:
Hello @dfy888 This error happens when a flowgraph from 3.8 or earlier is opened in 3.9 or newer. To fix the error just re-select the Cosine value in your signal source block.

duggabe, 2021-04-20:
@dfy888 : I have written a brief porting guide which addresses this issue: https://wiki.gnuradio.org/index.php/Porting_Existing_Flowgraphs_to_a_Newer_Version

dfy888, 2021-04-21:
> @dfy888 : I have written a brief porting guide which addresses this issue: https://wiki.gnuradio.org/index.php/Porting_Existing_Flowgraphs_to_a_Newer_Version

re-select "Hamming"   is not work too !

dfy888, 2021-04-21:
> Hello @dfy888 This error happens when a flowgraph from 3.8 or earlier is opened in 3.9 or newer. To fix the error just re-select the Cosine value in your signal source block.

not working!

willcode, 2021-05-01:
- Quit GRC
- `rm ~/.cache/grc_gnuradio/cache.json`
- Restart GRC
- Reselect Hamming (and other enums)