gnuradio/gnuradio #4548: CMAKE OOT Module: MPIR not found!

tayloreisman16, 2021-04-22:
Hey guys,

I've got OOT modules that I've needed to test, and this problem is something new to me that never appeared in previous versions of GNU Radio and Ubuntu.

If I try to use CMAKE on an OOT build folder, it reports missing the MPIR library. I've gone through source installs of MPIR and the downloadable tar.bz2 available on [MPIR's website](https://mpir.org/). I've tried the GNU Radio 3.9 branch on Ubuntu 20.04/18.04 with no luck. There are only two good sources mentioned on google for resolving these issues, both of which do not seem to work.

As this problem is repeatedly occurring now, I'm sure many of you have run into this issue. What is the general approach for linking the MPIR libraries to CMAKE correctly?

-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Found LOG4CPP: /usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   unit_test_framework
--   chrono
--   atomic
-- Found VOLK: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version "3.6.9") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable exact version "3.6.9") 
-- Found Git: /usr/bin/git  
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components:  doxygen missing components:  dot
CMake Error at CMakeLists.txt:122 (find_package):
  By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "pybind11",
  but CMake did not find one.

  Could not find a package configuration file provided by "pybind11" with any
  of the following names:

    pybind11Config.cmake
    pybind11-config.cmake

  Add the installation prefix of "pybind11" to CMAKE_PREFIX_PATH or set
  "pybind11_DIR" to a directory containing one of the above files.  If
  "pybind11" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/srv/LTE-GNU-Radio-Code/GNU-Radio-Repositories/TEST/gr-Test/build/CMakeFiles/CMakeOutput.log".
See also "/srv/LTE-GNU-Radio-Code/GNU-Radio-Repositories/TEST/gr-Test/build/CMakeFiles/CMakeError.log".

drmpeg, 2021-04-22:
You don't need MPIR. The dependency is being met with GMP. MPIR and GMP are the same thing, and CMake looks for both.
```
-- Checking for module 'gmp'
-- No package 'gmp' found
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so     <-------- GMP found here
-- Checking for module 'mpir >= 3.0'
-- No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR)
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so     <------- GMP resolves the dependency
```

Since folks get confused about this all the time, it's been updated recently.

https://github.com/gnuradio/gnuradio/pull/4414

The reason your source build of MPIR didn't work is because you have to use an extra flag to compile the C++ library.

You are missing pybind11. On Ubuntu 18.04, you have to build that from source. On 20.04, you can just apt install pybind11-dev.

tayloreisman16, 2021-04-23:
Thanks drmpeg, I'll try these out today!

closed 2021-04-23 by tayloreisman16