gnuradio/gnuradio #4552: The output of“constellation modulator”?

liumaolin-931, 2021-04-25:
Why is the output of ”constellation modulator” a complex signal? In general IQ modulation, the signals transmitted to the channel are not all real signals?

dave-pi, 2021-04-25:
It just makes the math easier to pack the two real baseband signals as a
complex value. It's easier to work with complex exponentials than with
products of sine and cosine functions. At the end you take real part of the
complex value, and that's what goes out 'in the air'.

liumaolin-931, 2021-04-26:
I see. Thank you very much for your reply!

dkozel, 2021-04-26:
Hi @liumaolin-931 I'm going to close this issue, but if you have any further questions please reopen it or join the chat at https://chat.gnuradio.org or the [email list](https://lists.gnu.org/mailman/listinfo/discuss-gnuradio).

closed 2021-04-26 by dkozel