gnuradio/gnuradio #4554: No GRC blocks or examples for gr-wavelet

mormj, 2021-04-28:
The gr-wavelet block have no GRC blocks or examples.  These would obviously be good to have to demonstrate the capability of gr-wavelet

mboros1, 2023-12-12:
I was looking into using a wavelet block and found they don't exist! Could I pick this issue up and maybe get some guidance on it?

willcode, 2023-12-12:
Sure. There are 3 blocks in the gr-wavelet module
- squash_ff
- wavelet_ff
- wvps_ff
If you take a look at the other modules, you will see grc subdirectories. The yaml files in the grc directories are what makes blocks available in GRC. There is also a CMakeLists.txt needed to specify which files get installed. The CMakeLists.txt in the top level directory of the module also says that the grc subdirectory should be processed.

I don't know much about the wavelet blocks, and I'm not sure who does, so after you get started, you might want to mention what you're doing in General Chat to see if anyone can help with examples or testing.

marcusmueller, 2024-01-28:
Yeah, not even the single qa_ file is in any way descriptive.

marcusmueller, 2024-01-28:
this code is effectively also untested, and without examples, we'll never know whether it works for *anyone*. this elevates this to a bug.

Aryanth29, 2026-11-08:
Hi, I have thought of a solution for this and am working on it!