gnuradio/gnuradio #4557: Wrong order in the generated .py causes error "object has no attribute 'tabs_layout_2'"

kaushiksv, 2021-04-30:
I am trying to run the example at https://github.com/bastibl/gr-rds/blob/3a80121f0858adf4ebd2f5f9f36ff1e4bb18b7a4/examples/rds_rx.grc (maint-3.9 branch)

I get the following error:

```
Executing: /usr/bin/python3.8 -u /home/kaushik/gr-rds/examples/rds_rx_kauzik.py

Traceback (most recent call last):
  File "/home/kaushik/gr-rds/examples/rds_rx_kauzik.py", line 625, in <module>
    main()
  File "/home/kaushik/gr-rds/examples/rds_rx_kauzik.py", line 603, in main
    tb = top_block_cls()
  File "/home/kaushik/gr-rds/examples/rds_rx_kauzik.py", line 131, in __init__
    self.tabs_layout_2.addWidget(self._qtgui_waterfall_sink_x_0_win)
  File "/usr/local/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 82, in __getattr__
    return getattr(self._impl, name)
AttributeError: 'gnuradio.gr.gr_python.top_block_pb' object has no attribute 'tabs_layout_2'

>>> Done (return code -11)
```
Problem seems to be the placement of the following section of code:
```
        self.tabs = Qt.QTabWidget()
        self.tabs_widget_0 = Qt.QWidget()
        self.tabs_layout_0 = Qt.QBoxLayout(Qt.QBoxLayout.TopToBottom, self.tabs_widget_0)
        self.tabs_grid_layout_0 = Qt.QGridLayout()
        self.tabs_layout_0.addLayout(self.tabs_grid_layout_0)
        self.tabs.addTab(self.tabs_widget_0, 'BB')
        self.tabs_widget_1 = Qt.QWidget()
        self.tabs_layout_1 = Qt.QBoxLayout(Qt.QBoxLayout.TopToBottom, self.tabs_widget_1)
        # And so on...
```
The member variables tabs\_layout\__[1-n]_ are being defined after the first reference. The user is forced to edit the .py file manually and run from terminal. 

GRC should place the such definitions before first reference.

GnuRadio maint-3.9. Built and installed from commit 44cd1949a027101f805233092e54b36978c55269.

daniestevez, 2021-05-02:
I was going to open a bug report of exactly the same thing. I'm using v3.9.1.0. Even the simplest example using tabs will generate code where `X_widget_n_layout_m` is used before it is created.

I'm replicating the problem with this flowgraph:
[image]
[test_tabs.grc.zip](https://github.com/gnuradio/gnuradio/files/6411408/test_tabs.grc.zip)

versd, 2021-05-10:
I have this bug too in 3.9.1.  I had to manually revert commit https://github.com/gnuradio/gnuradio/commit/f45c82436c2a19fa1e5ff15177fbc4cc591017cb to fix this error. Which might reintroduce the bug that commit tried to fix. Haven't done regression tests.

daniestevez, 2021-05-11:
I confirm that this workaround seems to work well with the test_tabs.grc file.
> As a workaround, add a gui hint like (0,0) to the tab widget and I think this won't happen.

 Thanks! I think that will save me some headaches until the bug gets fixed.

kaushiksv, 2021-05-11:
@ChristopheSeguinot  my mistake, sorry. It was supposed to be [this one (maint-3.8 in gr-rds)](https://raw.githubusercontent.com/bastibl/gr-rds/7d2198919f4f4bb7a1e9f5c03bfc7da4a2764bbe/examples/rds_rx.grc)

Thanks for the fix and also thanks to @versd  for pinpointing the commit.

willcode, 2021-05-12:
When a widget is placed in a tab, a hint (not really a hint) like `tab_name@0` is used. The ordering gets messed up because the widget, having a hint, gets built and placed in a tab that doesn't exist yet because the tab doesn't have a hint. Adding a hint to the tab gets things built in the right order, at least in simple cases. A little more ordering logic is needed to fix this.

PR #4609 (2021-06-01): grc: fix Wrong order in the generated .py caused by uncorrect function without_gui_hint(block)

This fix #4557 
I tested it under GR 3.9.1 using different flowgraph combining widgets with/without GUI_hint positioned Inside /outside tabs. 

Note: I =was not able to reproduce errror  in  #4557 first message ( https://github.com/bastibl/gr-rds/blob/3a80121f0858adf4ebd2f5f9f36ff1e4bb18b7a4/examples/rds_rx.grc (maint-3.9 branch) 
Thank to @kaushiksv for proposing this fix
closed 2021-06-01 by mormj