gnuradio/gnuradio #4558: Boost not found during cross-compilation

rodrirq, 2021-04-30:
Hello, I am trying to cross-compile GNU Radio for ARM.

I am using PetaLinux SDK cross-compilers.

These are my toolchain and the output of my cmake:

```
set( CMAKE_SYSTEM_NAME Linux )
#set( CMAKE_C_COMPILER  $ENV{CC} )
#set( CMAKE_CXX_COMPILER  $ENV{CXX} )
string(REGEX MATCH "sysroots/([a-zA-Z0-9]+)" CMAKE_SYSTEM_PROCESSOR $ENV{SDKTARGETSYSROOT})
string(REGEX REPLACE "sysroots/" "" CMAKE_SYSTEM_PROCESSOR ${CMAKE_SYSTEM_PROCESSOR})
set( CMAKE_CXX_FLAGS $ENV{CXXFLAGS}  CACHE STRING "" FORCE )
set( CMAKE_C_FLAGS $ENV{CFLAGS} CACHE STRING "" FORCE ) #same flags for C sources
set( CMAKE_LDFLAGS_FLAGS ${CMAKE_CXX_FLAGS} CACHE STRING "" FORCE ) #same flags for C sources
set( CMAKE_LIBRARY_PATH ${OECORE_TARGET_SYSROOT}/usr/lib )
set( CMAKE_FIND_ROOT_PATH $ENV{OECORE_TARGET_SYSROOT} $ENV{OECORE_NATIVE_SYSROOT} )
set( CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER )
set( CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY )
set( CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY )
set( ORC_INCLUDE_DIRS $ENV{OECORE_TARGET_SYSROOT}/usr/include/orc-0.4 )
set( ORC_LIBRARY_DIRS $ENV{OECORE_TARGET_SYSROOT}/usr/lib )
set( BOOST_INCLUDEDIR /opt/boost_1_76_0/include/boost-1_76/boost )
set( BOOST_LIBRARYDIR /opt/boost_1_76_0/lib )
```

```
$ cmake -Wno-dev -DCMAKE_TOOLCHAIN_FILE=/home/rodrirq/gnuradio/cmake/Toolchains/oe-sdk_cross.cmake -DENABLE_GR_WXGUI=OFF -DENABLE_GR_VOCODER=OFF -DENABLE_GR_DTV=OFF -DENABLE_GR_ATSC=OFF -DENABLE_DOXYGEN=OFF ../ 
-- The CXX compiler identification is GNU 9.2.0
-- The C compiler identification is GNU 9.2.0
-- Check for working CXX compiler: /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/bin/arm-xilinx-linux-gnueabi/arm-xilinx-linux-gnueabi-g++
-- Check for working CXX compiler: /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/bin/arm-xilinx-linux-gnueabi/arm-xilinx-linux-gnueabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/bin/arm-xilinx-linux-gnueabi/arm-xilinx-linux-gnueabi-gcc
-- Check for working C compiler: /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/bin/arm-xilinx-linux-gnueabi/arm-xilinx-linux-gnueabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Build type set to Release.
-- Found Git: /usr/bin/git  
-- Extracting version information from git describe...
-- Performing Test HAVE_VISIBILITY_HIDDEN
-- Performing Test HAVE_VISIBILITY_HIDDEN - Success
-- Performing Test HAVE_WARN_SIGN_COMPARE
-- Performing Test HAVE_WARN_SIGN_COMPARE - Success
-- Performing Test HAVE_WARN_ALL
-- Performing Test HAVE_WARN_ALL - Success
-- Performing Test HAVE_WARN_NO_UNINITIALIZED
-- Performing Test HAVE_WARN_NO_UNINITIALIZED - Success
-- Compiler Version: arm-xilinx-linux-gnueabi-gcc (GCC) 9.2.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/bin/arm-xilinx-linux-gnueabi/arm-xilinx-linux-gnueabi-gcc:::-O3 -DNDEBUG  -O2 -pipe -g -feliminate-unused-debug-types  -std=gnu99 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/bin/arm-xilinx-linux-gnueabi/arm-xilinx-linux-gnueabi-g++:::-O3 -DNDEBUG  -O2 -pipe -g -feliminate-unused-debug-types  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- Building Static Libraries: OFF
-- Could NOT find Boost: missing: date_time program_options filesystem system regex thread (found /opt/boost_1_76_0/lib/cmake/Boost-1.76.0/BoostConfig.cmake (found suitable version "1.76.0", minimum required is "1.48"))
-- Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS) (Required is at least version "2.7")
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Found SWIG: /usr/bin/swig3.0  
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = FALSE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Disabling python-support support.
--   Override with -DENABLE_PYTHON=ON/OFF
-- Found PkgConfig: /opt/pkg/sdk/sysroots/x86_64-petalinux-linux/usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'cppunit'
--   No package 'cppunit' found
-- Could NOT find CPPUNIT (missing: CPPUNIT_INCLUDE_DIRS) 
-- 
-- Configuring testing-support support...
--   Dependency CPPUNIT_FOUND = FALSE
--   Disabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
-- 
-- Configuring VOLK support...
-- Build type set to Release.
-- Extracting version information from git describe...
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- 
-- Python checking for python >= 2.7
-- Python checking for python >= 2.7 - found
-- 
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - found
-- 
-- Python checking for six - python 2 and 3 compatibility library
-- Python checking for six - python 2 and 3 compatibility library - found
-- Could NOT find Boost: missing: filesystem system (found /opt/boost_1_76_0/lib/cmake/Boost-1.76.0/BoostConfig.cmake (found suitable version "1.76.0", minimum required is "1.35"))
CMake Error at volk/CMakeLists.txt:112 (message):
  VOLK Requires boost to build

-- Configuring incomplete, errors occurred!
See also "/home/rodrirq/gnuradio/build/CMakeFiles/CMakeOutput.log".
```

As the cmake output says, I downloaded a boost_1_76_0. But even when I put the path into the toolchain, it doesn't get detected.

Can anyone help me?

balister, 2021-04-30:
Does the Petalinux SDK contain boost? I think you mention you added
boost, how did you add it?

Philip

On 4/30/21 4:53 AM, rodrirq wrote:
> Hello, I am trying to cross-compile GNU Radio for ARM.
>
> I am using PetaLinux SDK cross-compilers.
>
> These are my toolchain and the output of my cmake:
>
> ```
> set( CMAKE_SYSTEM_NAME Linux )
> #set( [...]

rodrirq, 2021-04-30:
Hello @balister,

Indeed. I downloaded boost, so my SDK shouldn't contain it.

How am I supposed to add it? 

Thank you for your answer.

rodrirq, 2021-04-30:
Hello @balister,

Indeed. I downloaded boost instead of adding it to the SDK. I have realised that when I do the petalinux-config -c rootfs, I can add boost. So I will do that and, if it works, I will close the issue.

Thank you for your answer.

rodrirq, 2021-04-30:
Yes, @balister.

Now it finds boost. Now after the cmake, I have this:

```
-- ######################################################
-- # Gnuradio enabled components                         
-- ######################################################
--   * volk
--   * sphinx
--   * gnuradio-runtime
--   * gr-ctrlport
--   * gr-blocks
--   * gr-fec
--   * gr-audio
--   * * oss
--   * gr-noaa
--   * gr-fcd
-- 
-- ######################################################
-- # Gnuradio disabled components                        
-- ######################################################
--   * python-support
--   * testing-support
--   * doxygen
--   * gnuradio-companion
--   * gr-fft
--   * gr-filter
--   * gr-analog
--   * gr-digital
--   * gr-dtv
--   * gr-atsc
--   * gr-comedi
--   * gr-channels
--   * gr-pager
--   * gr-qtgui
--   * gr-trellis
--   * gr-uhd
--   * gr-utils
--   * gr-video-sdl
--   * gr-vocoder
--   * gr-wavelet
--   * gr-wxgui
--   * gr-zeromq

```

If I want to work with gr-limesdr, do you know if I need any of the Gnuradio disabled components?

Thanks again.

PR #6318 (2022-11-07): cmake: Ensure that target names are unique
## Description
CMake sometimes fails with:
```
| CMake Error at cmake/Modules/GrPython.cmake:197 (add_custom_target):
|   add_custom_target cannot create target "pygen_gr_digital_examples_c8263"
|   because another target with the same name already exists.  The existing
|   target is a custom target created in source directory
|   "/home/buildbot/worker/lyon-worker/build-master-qemu-next/build/build/tmp-glibc/work/cortexa57-oe-linux/gnuradio/3.10.3.0+gitAUTOINC+3f9c56abc8-r0/git/gr-digital/examples".
|   See documentation for policy CMP0002 for more d [... cut]
closed 2021-05-03 by rodrirq