gnuradio/gnuradio #4560: Theme changes make no difference on MacOS Big Sur

NeuroForLunch, 2021-05-02:
Installed from source - everything works but theme changes make no difference even though config.conf gets updated with the selected theme:

```
[qtgui]
qss = /usr/local/Cellar/gnuradio/3.9.0.0_3/share/gnuradio/themes/dark.qss
style = raster
```

dkozel, 2021-05-07:
Hi @NeuroForLunch thanks for the report. I'm not able to test macOS, but possibly @michaelld may be able to.

NeuroForLunch, 2021-06-24:
I was able to get dark mode working by doing the following on OSX:

Create a file `~/.config/gtk-3.0/settings.ini` with the following content:
```
[Settings]
gtk-application-prefer-dark-theme=true
```

Find where the following files are installed and make changes:
```
grc/Core/Constants.py
grc/Core/platform.py
grc/gui/canvas/block.py
grc/gui/canvas/colors.py
grc/gui/canvas/connection.py
```

Same for Linux but the settings.ini file was not needed.

michaelld, 2021-09-25:
Hmmm ... I've never used themes but I love dark mode so I need to try this out! It would be really useful if we could just specify the theme as you note in the issue main message ... maybe under `[grc]` rather than `[qtgui]` but same basic concept.

michaelld, 2021-09-25:
There are some settings that work in GR config, and some that require GTK config ... like fonts. IDK why some work in some places but not in others. Guessing that's at least one part of the issue here.

marcusmueller, 2024-01-28:
This confuses two different things: the QSS theme for the Qt GUI you can generate, and the OS theme for GRC itself. So, luckily, no bug here. 

PS: the new qt GRC comes with a dark mode!

closed 2024-01-28 by marcusmueller