gnuradio/gnuradio #4561: Polyphase Arbitrary Resampler high CPU usage when throttled

daniestevez, 2021-05-02:
I have noticed that in certain circumstances the Polyphase Arbitrary Resampler block will have a high CPU usage (basically 100% of a CPU core) even when the data flow is limited by a throttle block.

The following flowgraph illustrates what I mean. There are throttle blocks that limit the rate to 2ksps, so the CPU usage should be very low.  If both throttles or only the throttle before the resampler are enabled, then the CPU usage is low as expected. However, if only the throttle after the resampler is enabled, the thread for the Polyphase Arbitrary Resampler will use up 100% of a CPU core. Moreover, the CPU usage of the Noise Source will also be high.

[image]
[test_arb_resampler.grc.zip](https://github.com/gnuradio/gnuradio/files/6411424/test_arb_resampler.grc.zip)

I haven't looked into the cause of this problem yet, but my impression is that the resampler `work()` function is called needlessly too often.

I'm using GNU Radio v3.9.1.0 for this test.

willcode, 2021-05-02:
Sounds like it's handling one sample at a time when rates are low, especially when interpolating.

daniestevez, 2021-05-02:
Even though very inefficient, that would be fine from the point of view of latency.

I feel that there is some busy waiting involved. Something like most of the calls to `work()` producing no samples. Even if I set the throttle rate to something crazy low like 10 sps I still get high CPU usage on the arbitrary resampler and noise source threads.

willcode, 2021-05-02:
If no taps are specified, `history()` is 0, so when the downstream is starved, `forecast()`
`ninput_items_required[i] = noutput_items / relative_rate() + history() - 1;`
could return 0. Or even -1?

willcode, 2021-05-02:
Resampling rate of 1.0 gives low CPU, 1.0001 gives 100%.

daniestevez, 2021-05-02:
I think that if you don't specify the taps, then a sinc prototype filter is created for the taps, but I can't see where in the code this happens.

In any case, the same happens if you specify taps (I originally found this problem when using the resampler with an RRC filter). For instance, in my example it is possible to set the taps to `32*3*[1]` to have a rectangular window of 3 samples. This also gives high CPU usage.

daniestevez, 2021-05-02:
Ah, found it! There's a [Python wrapper](https://github.com/gnuradio/gnuradio/blob/v3.9.1.0/gr-filter/python/filter/pfb.py#L193) that is used to create the taps if the user doesn't specify any.

marcusmueller, 2021-06-06:
This will get slightly under-prettified, but observations:

* 10% of CPU of the pfb_resampler thread (i.e. our DUT) is spent calling `clock_gettime`. Huh. I hope this is an instrumentation error.
* `int nitems = floorf((float)noutput_items / relative_rate());` we're pretty sure that's a non-negative number, so `int a = floor(b);` is the same as `int a = b;`, just way slower. Also, should be `unsigned int` while we're at it.

marcusmueller, 2021-06-06:
Yes, the high CPU usage comes from the overhead of doing small work calls. With perf counters disabled, the majority of gnuradio time is spent in `tpb_detail::notify_upstream`. With them enabled, that shifts toward the gettime call. Generally, condition variable waiting does also take also a lot of CPU.

Conclusion: hm. a bit better `forecast`ing might not hurt, avoiding such miniature calls to `general_work`. I wonder whether perfcounters *need* to be that CPU intense, or whether there would be better ways to measure time from user land. However, I believe instrumentation might better be done from eBPF, as @bastibl investigated in his paper on benchmarking GR scheduling.

daniestevez, 2021-06-11:
Hi Marcus,

Thanks for looking into this. What puzzles me is that even with a very low sample rate through the block, it still spends 100% of a CPU core, so it seems to me that the `work()` function is being called too many times without it performing any work. In other words, something is busy waiting for data instead of blocking.

I don't know much about how the scheduler works, but I seem to recall that if the work function didn't do any work then the scheduler wouldn't call it again in the next 100 ms to avoid this kind of busy waiting. I don't know if this applies in all situations, so perhaps this is what would be needed here?

willcode, 2021-07-14:
As an experiment, I `set_output_multiple()` - to something incorrect so I won't mention it - and the performance is great. Some combination of `forecast()` and `set_output_multiple()` mods may fix this. Note that `pfb_channelizer_ccf_impl.cc` contains
```
    // Calculate the number of filtering rounds to do to evenly
    // align the input vectors with the output channels
    d_output_multiple = 1;
    while ((d_output_multiple * d_rate_ratio) % d_nfilts != 0)
        d_output_multiple++;
    set_output_multiple(d_output_multiple);
```
in the constructor. Something similar might work.

willcode, 2021-07-14:
@marcusmueller @daniestevez 
When rate out > rate in, the scheduler needs a little extra help. This seems to work:
```
    unsigned multiple = 1;
    if (rate > 1.0f) {
      multiple = std::max<int>(rate, filter_size);
    }
    set_output_multiple(multiple);
```
This makes it so that at least 1 sample can be processed on the input. `filter_size` is really the number of PFB filter arms, which is a sort of arbitrary but reasonable minimum number of samples to output.

I tried the original flowgraph with
- 2000 * 1.001
- 2000 * 1000
- 2000000 * 0.001
and in each case switched the throttle between upstream and downstream of the filter. CPU usage was "normal".

If there is not anything obviously wrong, I'll make these changes in the right places and submit a PR.

daniestevez, 2021-07-16:
Thanks for taking the time to look into this! I'm still trying to wrap my head around how/why your trick works. It seems good enough to fix the problem in most use cases, particularly in the one which lead me to finding this problem: a PSK narrowband transmitter that should run a rather low sample rate, and so spend very little CPU time.

willcode, 2021-07-16:
I think it causes `BLKD_OUT` here at `block_executor.cc:348`.
```
        // determine the minimum available output space
        noutput_items =
            min_available_space(d, m->output_multiple(), m->min_noutput_items());
        if (ENABLE_LOGGING) {
            std::ostringstream msg;
            msg << "regular ";
            msg << m->relative_rate_i() << ":" << m->relative_rate_d();
            msg << "  max_items_avail = " << max_items_avail;
            msg << "  noutput_items = " << noutput_items;
            GR_LOG_INFO(d_debug_logger, msg.str());
        }
        if (noutput_items == -1) // we're done
            goto were_done;

        if (noutput_items == 0) { // we're output blocked
            LOG(GR_LOG_INFO(d_debug_logger, "BLKD_OUT"););
            return BLKD_OUT;
        }
```
The "fix" puts a lower bound on how much data the block produces per iteration when upsampling. I didn't want to change anything in the scheduler so it came down to what could be changed in the block that had an effect. When downsampling, some backpressure is created where there would be no samples produced, and that is apparently enough to keep the scheduler from doing a busy wait with the resampler block.

closed 2021-09-02 by mormj