gnuradio/gnuradio #4571: soapy: API completeness

ncorgan, 2021-05-04:
With the intent of matching the granularity of gr-uhd:

- [ ] Getting device info
- [ ] Getters are protected in block_impl instead of public for block (not sure why), should be exposed
- [ ] Some missing functions

willcode, 2021-05-04:
Yes, I was thinking of adding more functions, but did not expose anything that was not exposed in the original code, for the first round. Many users don't need anything more than the current API, so this gets the feature out and tested a bit. The getters should be an easy one.

One thing I want to put some thought into is how interactive things should be. There could even be a fully interactive QT Soapy SDR block, like the Soapy blocks in Pothos, but not sure yet.

ncorgan, 2021-05-04:
I've been knee-deep in Soapy recently, so go ahead and assign this to me. I can get PRs up for some of this over the next few days.

willcode, 2021-05-04:
Cool - we can coordinate, no need for a specific assignment. Thanks!

We also need to dup the Custom Soapy Source/Sink logic out of the original yml files to support any 'ole device, command line specification of devices, etc.

willcode, 2021-05-08:
@ncorgan anything left for this issue?

ncorgan, 2021-05-08:
The checkboxes in the issue description are still valid. The most recent branch added everything needed for the 0.7.2 features currently exposed by the block, but I'd like to add the remaining 0.7.2 functionality appropriate for a block. I see this as two more branches:
 * The remaining functions needing the new custom pybind11, as there's a minor refactor there.
 * The rest of it, which is just thin wrappers around the `SoapySDR::Device` functions.

willcode, 2021-05-15:
@ncorgan what do you have in mind for additional messages?

ncorgan, 2021-05-15:
gr-uhd has messages for most setters, so I have a WIP branch with similar.

willcode, 2021-05-20:
I'd previously shortened a couple of the message names, but I'll change them back so they match the uhd ones where applicable (bandwidth, antenna, ...). We don't have any kind of "standard" but might as well match what's there.

closed 2021-07-12 by mormj