gnuradio/gnuradio #4572: Opening GNURadio-Companion

rolledankle1000, 2021-05-04:
Hello All, I have ran into an issue trying to open GNURadio-Companion. I have downloaded GNU Radio via the website link seen below. 
https://wiki.gnuradio.org/index.php/InstallingGR#Step_3:_Installing_GNU_Radio
And from what I can tell, It has downloaded and installed successfully, as I got no errors while downloading. It was installed on Ubuntu 18.04 LTS. When trying to open the GUI, I use the command "gnuradio-companion" as shown on the various websites and manuals, but I get the following error. 

rams_raiders@E5-EE-E201LT5:~$ python3 -V
Python 3.7.10
rams_raiders@E5-EE-E201LT5:~$ gnuradio-companion
Unable to init server: Could not connect: Connection refused
Unable to init server: Could not connect: Connection refused
Unable to init server: Could not connect: Connection refused
/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp, stat
Traceback (most recent call last):
  File "/usr/local/bin/gnuradio-companion", line 102, in <module>
    run_main()
  File "/usr/local/bin/gnuradio-companion", line 95, in run_main
    exit(main())
  File "/usr/local/lib/python3/dist-packages/gnuradio/grc/main.py", line 73, in main
    from .gui.Platform import Platform
  File "/usr/local/lib/python3/dist-packages/gnuradio/grc/gui/Platform.py", line 26, in <module>
    from . import canvas
  File "/usr/local/lib/python3/dist-packages/gnuradio/grc/gui/canvas/__init__.py", line 21, in <module>
    from .param import Param
  File "/usr/local/lib/python3/dist-packages/gnuradio/grc/gui/canvas/param.py", line 23, in <module>
    from .. import ParamWidgets, Utils, Constants
  File "/usr/local/lib/python3/dist-packages/gnuradio/grc/gui/ParamWidgets.py", line 69, in <module>
    add_style_provider()
  File "/usr/local/lib/python3/dist-packages/gnuradio/grc/gui/ParamWidgets.py", line 67, in add_style_provider
    Gtk.STYLE_PROVIDER_PRIORITY_APPLICATION
TypeError: Argument 0 does not allow None as a value

Is there anyone that knows how to fix this? Also shown in my current version of python3. Thanks!

mormj, 2021-05-04:
Are you executing gnuradio-companion on a remote server or is it the local machine?

rolledankle1000, 2021-05-04:
> Are you executing gnuradio-companion on a remote server or is it the local machine?

It is on the local machine, As in my ubuntu terminal on my laptop

mormj, 2021-05-04:
You may get more folks who have seen something similar by describing this issue on the discuss-gnuradio mailing list - https://lists.gnu.org/mailman/listinfo/discuss-gnuradio

dkozel, 2021-05-07:
@rolledankle1000 "ubuntu terminal on my laptop" Is this a Windows Subsystem for Linux terminal? That's where I've usually seen that message. If so I recently added some additional information to our install guide.
https://wiki.gnuradio.org/index.php/WindowsInstall#WSL_.7C_Ubuntu

mbr0wn, 2021-05-14:
@rolledankle1000 thanks for trying GNU Radio. This error is a generic permission error (no access to the X-server) and is not GNU Radio-related. Try opening another graphical application and see if that trips  the error as well.

closed 2021-05-14 by mbr0wn

rolledankle1000, 2021-05-18:
> @rolledankle1000
> 
> > ubuntu terminal on my laptop
> 
> Is this a Windows Subsystem for Linux terminal? That's where I've usually seen that message. If so I recently added some additional information to our install guide.
> https://wiki.gnuradio.org/index.php/WindowsInstall#WSL_.7C_Ubuntu

This is a windows subsystem of linux. Such as Ive downloaded Ubuntu from the microsoft store app

rolledankle1000, 2021-05-18:
> This is a windows subsystem of linux. Such as Ive downloaded Ubuntu from the microsoft store app

> @rolledankle1000
> 
> > ubuntu terminal on my laptop
> 
> Is this a Windows Subsystem for Linux terminal? That's where I've usually seen that message. If so I recently added some additional information to our install guide.
> https://wiki.gnuradio.org/index.php/WindowsInstall#WSL_.7C_Ubuntu

Your advice worked  Thanks!