gnuradio/gnuradio #4595: qa_agc AssertionError: 96 != 100

MountainLogic, 2021-05-05:
I'm seeing only one qa error.  I've used ctest -V to generate more detailed debug info.
This was built on a fresh out of the shrink wrap Dell XPS 13 9310 with factory install Ubuntu 20.04.  This has has an 11th Gen  i7-1185G7 @ 3.00GHz x 8.  I hope Intel has not installed an upgraded version of the old floating point ;-)
I followed the generic [prereqs](https://wiki.gnuradio.org/index.php/UbuntuInstall#Focal_Fossa_.2820.04.29) for Ubuntu 20.04 and followed the [generic instructions](https://wiki.gnuradio.org/index.php/InstallingGR#From_Source) from source.

```
148/241 Test #148: qa_single_pole_iir ...........................   Passed    0.18 sec
test 149
        Start 149: qa_agc

149: Test command: /usr/bin/sh "/home/scott/wip/gnuradio/build/gr-analog/python/analog/qa_agc_test.sh"
149: Test timeout computed to be: 10000000
149: .........F..
149: ======================================================================
149: FAIL: test_006 (__main__.test_agc)
149: Test the complex AGC loop (attack and decay rate inputs)
149: ----------------------------------------------------------------------
149: Traceback (most recent call last):
149:   File "/home/scott/wip/gnuradio/gr-analog/python/analog/qa_agc.py", line 474, in test_006
149:     self.assertFloatTuplesAlmostEqual(result, M * [ref, ], 4)
149:   File "/home/scott/wip/gnuradio/gnuradio-runtime/python/gnuradio/gr_unittest.py", line 99, in assertFloatTuplesAlmostEqual
149:     self.assertEqual(len(a), len(b))
149: AssertionError: 96 != 100
149: 
149: ----------------------------------------------------------------------
149: Ran 12 tests in 0.008s
149: 
149: FAILED (failures=1)
149/241 Test #149: qa_agc .......................................***Failed    0.20 sec
test 150
        Start 150: qa_cpfsk

```

dkozel, 2021-05-07:
Hello Scott, Can you confirm which version of GNU Radio you've built? Since it's from source are you able to share the git hash?

Thanks!

MountainLogic, 2021-05-07:
@dkozel 

I believe that I built with 
git rev-parse master
930b332e70884bc57dc57b15294bbc2a7548fa3b

I'm also playing with git rev-parse feature/gr-iio and seeing the same assertion error.
git rev-parse feature/gr-iio
b13303bd8f858f312cf13ec67708d38976113aca

willcode, 2021-05-11:
From your build dir, can you run `gr-analog/python/analog/qa_agc_test.sh` several times and see if the output is the same? This looks more like "GR's brakes don't work that well" problem, not an actual problem with the block (or the processor).

MountainLogic, 2021-05-11:
@willcode , I consistently get  the same failure from your requested script:
```
scott@scott:~/wip/gnuradio/build$ gr-analog/python/analog/qa_agc_test.sh
.........F..
======================================================================
FAIL: test_006 (__main__.test_agc)
Test the complex AGC loop (attack and decay rate inputs)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/scott/wip/gnuradio/gr-analog/python/analog/qa_agc.py", line 474, in test_006
    self.assertFloatTuplesAlmostEqual(result, M * [ref, ], 4)
  File "/home/scott/wip/gnuradio/gnuradio-runtime/python/gnuradio/gr_unittest.py", line 99, in assertFloatTuplesAlmostEqual
    self.assertEqual(len(a), len(b))
AssertionError: 96 != 100

----------------------------------------------------------------------
Ran 12 tests in 0.012s

FAILED (failures=1)

```

Note that I had to `sudo make install` before I could run the test

MountainLogic, 2021-05-13:
Thanks @willcode .  I'll not bother shorting Intel stock then ;-)

mbr0wn, 2021-05-14:
I've seen this on our CI as well.

mbr0wn, 2021-05-14:
...and we have currently disabled it.

pitchea, 2021-06-10:
Hello, I figured I would add to this open issue. I am also seeing a similar one qa error for agc. I've also provided my  `ctest -V -R qa_agc` multiple time to generate more detailed debug info.

```
UpdateCTestConfiguration  from :/home/neudose/github/gnuradio/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/neudose/github/gnuradio/build/DartConfiguration.tcl
Test project /home/neudose/github/gnuradio/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 280
    Start 280: qa_agc

280: Test command: /bin/sh "/home/neudose/github/gnuradio/build/gr-analog/python/analog/qa_agc_test.sh"
280: Test timeout computed to be: 9.99988e+06
280: .........F..
280: ======================================================================
280: FAIL: test_006 (__main__.test_agc)
280: Test the complex AGC loop (attack and decay rate inputs)
280: ----------------------------------------------------------------------
280: Traceback (most recent call last):
280:   File "/home/neudose/github/gnuradio/gr-analog/python/analog/qa_agc.py", line 486, in test_006
280:     self.assertFloatTuplesAlmostEqual(result, M*[ref,], 4)
280:   File "/home/neudose/github/gnuradio/gnuradio-runtime/python/gnuradio/gr_unittest.py", line 110, in assertFloatTuplesAlmostEqual
280:     self.assertEqual(len(a), len(b))
280: AssertionError: 96 != 100
280: 
280: ----------------------------------------------------------------------
280: Ran 12 tests in 0.009s
280: 
280: FAILED (failures=1)
280: DEPRECATED: Using filename with gr_unittest does no longer have any effect.
1/1 Test #280: qa_agc ...........................***Failed    0.52 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.53 sec

The following tests FAILED:
	280 - qa_agc (Failed)
Errors while running CTest
```

I am building GNURadio on a brand new Dell Inspiron 15 5501 with a fresh install of Ubuntu 18.04 LTS. It also has the latest intel 11th Gen i7 1165G1 @ 2.8 GHz x8. I am installing GNURadio on the branch `maint-3.8`

I have also run `gr-analog/python/analog/qa_agc_test.sh` from the build directory
```
DEPRECATED: Using filename with gr_unittest does no longer have any effect.
.........F..
======================================================================
FAIL: test_006 (__main__.test_agc)
Test the complex AGC loop (attack and decay rate inputs)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/neudose/github/gnuradio/gr-analog/python/analog/qa_agc.py", line 486, in test_006
    self.assertFloatTuplesAlmostEqual(result, M*[ref,], 4)
  File "/home/neudose/github/gnuradio/gnuradio-runtime/python/gnuradio/gr_unittest.py", line 110, in assertFloatTuplesAlmostEqual
    self.assertEqual(len(a), len(b))
AssertionError: 96 != 100

----------------------------------------------------------------------
Ran 12 tests in 0.009s

FAILED (failures=1)
```

shuimoshusheng, 2021-07-05:
Has this problem been solved? I have same problem like this on ubuntu 18.04 gnuradio maint-3.8
root@john:~/gnuradio/build# ctest -V -R qa_agc
UpdateCTestConfiguration  from :/root/gnuradio/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/root/gnuradio/build/DartConfiguration.tcl
Test project /root/gnuradio/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 280
    Start 280: qa_agc

280: Test command: /bin/sh "/root/gnuradio/build/gr-analog/python/analog/qa_agc_test.sh"
280: Test timeout computed to be: 9.99988e+06
280: .........F..
280: ======================================================================
280: FAIL: test_006 (__main__.test_agc)
280: Test the complex AGC loop (attack and decay rate inputs)
280: ----------------------------------------------------------------------
280: Traceback (most recent call last):
280:   File "/root/gnuradio/gr-analog/python/analog/qa_agc.py", line 486, in test_006
280:     self.assertFloatTuplesAlmostEqual(result, M*[ref,], 4)
280:   File "/root/gnuradio/gnuradio-runtime/python/gnuradio/gr_unittest.py", line 110, in assertFloatTuplesAlmostEqual
280:     self.assertEqual(len(a), len(b))
280: AssertionError: 96 != 100
280: 
280: ----------------------------------------------------------------------
280: Ran 12 tests in 0.008s
280: 
280: FAILED (failures=1)
280: DEPRECATED: Using filename with gr_unittest does no longer have any effect.
1/1 Test #280: qa_agc ...........................***Failed    0.47 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.47 sec

The following tests FAILED:
	280 - qa_agc (Failed)
Errors while running CTest
root@john:~/gnuradio/build#

willcode, 2021-07-07:
No, the test still has trouble and is disabled for github testing:

`cd /build && ctest --output-on-failure -E "qa_agc|qa_cpp_py_binding|qa_cpp_py_binding_set|qa_ctrlport_probes|qa_qtgui"`

You can do your testing with the same tests suppressed if you need to.