gnuradio/gnuradio #4600: Time Sink Trigger Auto Mode Bug?

Jia-Yin, 2021-05-07:
It seems that the QT GUI Time Sink has wrong operation for the trigger mode set to auto.

According to the documentation:
... auto will trigger if the trigger event is seen, but will still plot otherwise, 
and normal will hold until the trigger event is observed.

I think it means that even there is no trigger event found, the scope will also plot the signal.
And this is what a real scope does.

However, just like the attached file,
I found that there is no signal shown in the scope if the trigger event is not met.
The signal will be shown only when the amplitude is large than 1.1 or when the trigger mode set to free. The auto mode just has the same effect with normal mode.

Is it a bug or I misunderstand its meaning?

[trigger_auto_mode.zip](https://github.com/gnuradio/gnuradio/files/6441468/trigger_auto_mode.zip)

dkozel, 2021-05-14:
From https://uk.tek.com/document/online/primer/xyzs-scopes/ch4/oscilloscope-systems-and-controls#C4section4

> In normal mode the oscilloscope only sweeps if the input signal reaches the set trigger point. Otherwise, the screen is blank (on an analog oscilloscope) or frozen (on a digital oscilloscope) on the last acquired waveform. Normal mode can be disorienting since you may not see the signal at first if the level control is not adjusted correctly.
> Auto mode causes the oscilloscope to sweep, even without a trigger. If no signal is present, a timer in the oscilloscope triggers the sweep. This ensures that the display will not disappear if the signal does not cause a trigger.

Looking quickly at the code this does seem like normal and auto have reasonable implementations.
https://github.com/gnuradio/gnuradio/blob/9e32ef0ffab29efe4c52f2e574016cd3d6453484/gr-qtgui/lib/time_sink_c_impl.cc#L492-L498

I haven't had a chance to check if does actually work correctly.

dkozel, 2021-05-14:
An auto trigger will happen every `Time Sink X length` buffers. In your case that's 1024 buffers, and at 32000 samples / 1024 samples / buffer that's once every 32 seconds! At 1 MS/s you'd get 1 forced trigger per second. The better behavior would probably be to trigger once or twice a second at minimum (as long as samp_rate > number of samples per time display).

closed 2024-02-14 by marcusmueller