gnuradio/gnuradio #4601: SoapSDR::loadModule error

cyprysp, 2021-05-08:
I was trying to build a simple project with osmocom Source, DCBlocker and QU GUI Sink, and I have trouble running it on hackrf one. 
Error: 

`[ERROR] SoapySDR::loadModule(/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so)
  dlopen() failed: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so: undefined symbol: _ZN5boost6chrono12steady_clock3nowEv
Using HackRF One with firmware local-57ed3eb
Traceback (most recent call last):
  File "./defaultawd.py", line 171, in <module>
    main()
  File "./defaultawd.py", line 149, in main
    tb = top_block_cls()
  File "./defaultawd.py", line 121, in __init__
    self.connect((self.osmosdr_source_0, 0), (self.dc_blocker_xx_0, 0))
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 37, in wrapped
    func(self, src, src_port, dst, dst_port)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 100, in connect
    self.primitive_connect(*args)
TypeError: primitive_connect(): incompatible function arguments. The following argument types are supported:
    1. (self: gnuradio.gr.gr_python.hier_block2_pb, block: gnuradio.gr.gr_python.basic_block) -> None
    2. (self: gnuradio.gr.gr_python.hier_block2_pb, src: gnuradio.gr.gr_python.basic_block, src_port: int, dst: gnuradio.gr.gr_python.basic_block, dst_port: int) -> None

Invoked with: <gnuradio.gr.gr_python.top_block_pb object at 0x7f28a7731d70>, <Swig Object of type 'gr::basic_block_sptr *' at 0x7f28a4a72f30>, 0, <gnuradio.gr.gr_python.basic_block object at 0x7f28a7038270>, 0
swig/python detected a memory leak of type 'gr::basic_block_sptr *', no destructor found.
`

During generation:

`Executing: /usr/bin/python3 -u /home/bilodong/hackrf/defaultawd.py

gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
[1m[31m[ERROR] SoapySDR::loadModule(/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so)
  dlopen() failed: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so: undefined symbol: _ZN5boost6chrono12steady_clock3nowEv[0m
Using HackRF One with firmware local-57ed3eb
Traceback (most recent call last):
  File "/home/bilodong/hackrf/defaultawd.py", line 171, in <module>
    main()
  File "/home/bilodong/hackrf/defaultawd.py", line 149, in main
    tb = top_block_cls()
  File "/home/bilodong/hackrf/defaultawd.py", line 121, in __init__
    self.connect((self.osmosdr_source_0, 0), (self.dc_blocker_xx_0, 0))
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 37, in wrapped
    func(self, src, src_port, dst, dst_port)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 100, in connect
    self.primitive_connect(*args)
TypeError: primitive_connect(): incompatible function arguments. The following argument types are supported:
    1. (self: gnuradio.gr.gr_python.hier_block2_pb, block: gnuradio.gr.gr_python.basic_block) -> None
    2. (self: gnuradio.gr.gr_python.hier_block2_pb, src: gnuradio.gr.gr_python.basic_block, src_port: int, dst: gnuradio.gr.gr_python.basic_block, dst_port: int) -> None

Invoked with: <gnuradio.gr.gr_python.top_block_pb object at 0x7f4ff41456f0>, <Swig Object of type 'gr::basic_block_sptr *' at 0x7f4ff1a97bd0>, 0, <gnuradio.gr.gr_python.basic_block object at 0x7f4ff5df3830>, 0
swig/python detected a memory leak of type 'gr::basic_block_sptr *', no destructor found.`

willcode, 2021-05-08:
This is most likely an incompatibility between versions of SoapyHackRF and the hackrf library, and not related to GNU Radio.

willcode, 2021-05-08:
Or SoapySDR vs Boost. Closing - not a GNU Radio issue.

closed 2021-05-08 by willcode