gnuradio/gnuradio #4607: Pathing issues

rodrirq, 2021-05-11:
Hello, I have installed GNU Radio in /gnuradio.

I am trying to install gr-limesdr but, all the includes have a bad pathing.

For example, it can't find #include <gnuradio/api.h>, I have to set the full path as <gnuradio/include/gnuradio/api.h>.

But, if I have to do this for each include, it would be crazy.

Is there any way to change this? Thank you in advance.

solomonbstoner, 2021-05-11:
Try steps A and D in https://wiki.gnuradio.org/index.php/ModuleNotFoundError . I believe it should resolve your path problem.

mormj, 2021-05-11:
When gnuradio is not installed under the normal search path such as `\usr\`, you have to tell the OOT module where to find it.  It looks like the OOT is looking a couple levels up from where it should.

When building the OOT, you have to:
```
mkdir build
cd build
cmake .. -DCMAKE_INSTALL_PREFIX=$PREFIX
```
where `$PREFIX` is the directory that holds the installed `include`, `lib`, etc. dirs

mormj, 2021-05-11:
Closing this as a github issue since it is not a bug with GNU Radio, or something to be addressed in the GR codebase, and it is being appropriately discussed on chat.gnuradio.org.

closed 2021-05-11 by mormj

rodrirq, 2021-05-11:
@mormj I installed GNU Radio at /gnuradio. I have tried putting /gnuradio as the DCMAKE_INSTALL_PREFIX of gr-limesdr, where it is installed the include, lib, etc, libexec, bin and share. But I get the same error.

GNU Radio is installed by default at /usr/local.

rodrirq, 2021-05-12:
@solomonbstoner I installed it there because I had more apps in /usr/local and I thought it was a good idea. I see that it wasn't. 

I will reinstall GNU Radio in /usr/local to avoid errors. 

Thank you for your answer.

rodrirq, 2021-05-12:
@solomonbstoner I reinstalled GNU Radio in its default directory, which is /usr/local. But when I try to install the gr-limesdr, I get the same error during the make.

solomonbstoner, 2021-05-13:
I got  an email about your reply, but I dont see it here. Have you managed to resolve your matter @rodrirq ?

rodrirq, 2021-05-14:
Hello @solomonbstoner

I deleted it because i solved that. I managed to resolve the problem with the includes, but it had nothing to do with my path.

math.h called api.h as <gnuradio/api.h>, when math.h and api.h are in the same folder, so I had to create that path by myself.

The problem I have now is ModuleNotFoundError, which I am trying to investigate right now.

Thank you for your answer.