gnuradio/gnuradio #4610: Using GNU Radio on PetaLinux

rodrirq, 2021-05-13:
Hello, I have successfully installed GNU Radio on PetaLinux.

The thing is that I have no idea on what I can do with it.

Is it possible to work coding? Because I only see people working with blocks.

I have looked for tutorials or something, but I don't find anything.

I tried to run the ./gnuradio-config-info and it works. But when I run ./gnuradio-companion or another program, I get the ModuleNotFoundError.

I hope someone could help me, thank you.

mormj, 2021-05-13:
@rodrirq - please direct usage questions like this to the chat (chat.gnuradio.org) or to the mailing list - https://wiki.gnuradio.org/index.php/MailingLists

closed 2021-05-13 by mormj

rodrirq, 2021-05-13:
@mormj I tried to use the mailing list. But I requested a subscription but when I try to log, it says that I am not registered.

Is it possible to send e-mails to discuss-gnuradio@gnu.org even when I get this error?

I have discussed with people of the chat, but I didn't get the correct answer.

mormj, 2021-05-13:
No, emails get blocked from non-registered users. You could make a gmail acct just for the list if whatever domain you had tried to register isn't working.

rodrirq, 2021-05-17:
@mormj I subscribed and got this:
```
Discuss-gnuradio Subscription results
Your subscription request has been received, and will soon be acted upon. Depending on the configuration of this mailing list, your subscription request may have to be first confirmed by you via email, or approved by the list moderator. If confirmation is required, you will soon get a confirmation email which contains further instructions.
```
I got the same message last time, but I never got accepted.
Any idea?