gnuradio/gnuradio #4613: gnuradio compilation error

aishwarya418112, 2021-05-14:
Hi, I am using UBUNTU 20.04 and am trying to install gnuradio 3.8. I have purged the 3.9 v of gnuradio from my system and was trying to build 3.8 v from source when I get the following errors....as for python, i installed 2.7 v seeing the error, but it is not able to recognise the version.below is the log of the terminal

cmake -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE=/usr/bin/python2.7 ../
-- Build type set to Release.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring testing-support support...
--   Enabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
-- Extracting version information from git describe...
-- Compiler Version: cc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- User set python executable /usr/bin/python2.7
-- Found PythonInterp: /usr/bin/python2.7 (found suitable version "2.7.18", minimum required is "2.7.6") 
-- Could NOT find PythonLibs: Found unsuitable version "3.8.5", but required is exact version "2.7" (found /usr/lib/x86_64-linux-gnu/libpython3.8.so)
-- Python checking for six - python 2 and 3 compatibility library - found
-- 
-- Checking for module SWIG
-- Found SWIG version 4.0.1.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = FALSE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Dependency SIX_FOUND = TRUE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled python-support but configuration checked failed
Call Stack (most recent call first):
  CMakeLists.txt:381 (GR_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/home/username/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/home/username/gnuradio/build/CMakeFiles/CMakeError.log".

Please help me solve this!

dkozel, 2021-05-14:
Hello @aishwarya418112 Do you need to use Python 2 for a specific reason, or would Python 3 be ok?

Do you need to build from source or would installing with the PPA be ok?
https://wiki.gnuradio.org/index.php/InstallingGR#Ubuntu_PPA_Installation

mbr0wn, 2021-05-14:
Hey @aishwarya418112, please don't post support questions here. We are happy to help you with these questions on the mailing list.

Hint: Try installing python3-dev or python2-dev or something like that.

closed 2021-05-14 by mbr0wn

aishwarya418112, 2021-05-16:
Hi,
    Actually the gnuradio version compatible with osmosdr driver is 3.8 and
so for that I need python 2.7 which I had installed, but it was unable to
recognise the path where the 2.7 v is I suppose and hence it is throwing
that error. I don't understand what to do anymore, so I am trying to
install a fresh ubuntu 18.04 instead of the 20.04 I now have and start
everything afresh. Any ideas though on how  I can use the 20.04 still.?