gnuradio/gnuradio #4615: Interleave / Deinterleave GRC block: inconsistent naming of parameters

marcusmueller, 2021-05-14:
These blocks use "Vec Length" instead of "vector length" AND this is not documented on the [wiki page](https://wiki.gnuradio.org/index.php/Deinterleave)s. It's really just a multiplier for the itemsize.

To fix this, 

1. Fix the GRC .block.yml files to use the canonical "vector size" / "vector length" (look that up)
2. Modify the wiki pages for the deinterleave / interleave blocks, explaining what the effect actually is, maybe make a figure / drawing to understand what the interleaver does when you set vector size.

solomonbstoner, 2021-05-18:
I notice sometimes "vector size" is used to refer to "vector length". Shall I change em to "vector length" or is "vector size" ok? Is the goal of this issue just to standardize "vec" to "vector" in the documentation?

[image]

solomonbstoner, 2021-05-20:
We're left with part 2 of this issue:

> Modify the wiki pages for the deinterleave / interleave blocks, explaining what the effect actually is, maybe make a figure / drawing to understand what the interleaver does when you set vector size >1.

closed 2021-09-23 by marcusmueller