gnuradio/gnuradio #4619: pmt::pmt_*vector_elements() functions not exported

ncorgan, 2021-05-14:
Adding a call to one of these functions to the QA test results in the following:
```
[100%] Building CXX object gnuradio-runtime/lib/CMakeFiles/pmt_unv.dir/pmt/qa_pmt_unv.cc.o
[100%] Linking CXX executable pmt_unv
/usr/bin/ld: CMakeFiles/pmt_unv.dir/pmt/qa_pmt_unv.cc.o: in function `test_u8vector::test_method()':
qa_pmt_unv.cc:(.text+0x1052c): undefined reference to `pmt::pmt_u8vector_elements(std::shared_ptr<pmt::pmt_base>)'
collect2: error: ld returned 1 exit status
make[2]: *** [gnuradio-runtime/lib/CMakeFiles/pmt_unv.dir/build.make:96: gnuradio-runtime/lib/pmt_unv] Error 1
make[1]: *** [CMakeFiles/Makefile2:4567: gnuradio-runtime/lib/CMakeFiles/pmt_unv.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

Confirming with `nm`:
```
nm -C pmt/libgnuradio-pmt.so | grep ' T .*vector_elem'
00000000000317a0 T pmt::s8vector_elements(std::shared_ptr<pmt::pmt_base>)
000000000002cbf0 T pmt::s8vector_elements(std::shared_ptr<pmt::pmt_base>, unsigned long&)
00000000000315e0 T pmt::u8vector_elements(std::shared_ptr<pmt::pmt_base>)
000000000002c720 T pmt::u8vector_elements(std::shared_ptr<pmt::pmt_base>, unsigned long&)
0000000000031220 T pmt::c32vector_elements(std::shared_ptr<pmt::pmt_base>)
0000000000030600 T pmt::c32vector_elements(std::shared_ptr<pmt::pmt_base>, unsigned long&)
0000000000031400 T pmt::c64vector_elements(std::shared_ptr<pmt::pmt_base>)
0000000000030f40 T pmt::c64vector_elements(std::shared_ptr<pmt::pmt_base>, unsigned long&)
0000000000032390 T pmt::f32vector_elements(std::shared_ptr<pmt::pmt_base>)
000000000002f3c0 T pmt::f32vector_elements(std::shared_ptr<pmt::pmt_base>, unsigned long&)
0000000000032550 T pmt::f64vector_elements(std::shared_ptr<pmt::pmt_base>)
000000000002fc80 T pmt::f64vector_elements(std::shared_ptr<pmt::pmt_base>, unsigned long&)
0000000000031ae0 T pmt::s16vector_elements(std::shared_ptr<pmt::pmt_base>)
000000000002d600 T pmt::s16vector_elements(std::shared_ptr<pmt::pmt_base>, unsigned long&)
0000000000031e50 T pmt::s32vector_elements(std::shared_ptr<pmt::pmt_base>)
000000000002e080 T pmt::s32vector_elements(std::shared_ptr<pmt::pmt_base>, unsigned long&)
00000000000321d0 T pmt::s64vector_elements(std::shared_ptr<pmt::pmt_base>)
000000000002eaa0 T pmt::s64vector_elements(std::shared_ptr<pmt::pmt_base>, unsigned long&)
0000000000031930 T pmt::u16vector_elements(std::shared_ptr<pmt::pmt_base>)
000000000002d0c0 T pmt::u16vector_elements(std::shared_ptr<pmt::pmt_base>, unsigned long&)
0000000000031c90 T pmt::u32vector_elements(std::shared_ptr<pmt::pmt_base>)
000000000002db40 T pmt::u32vector_elements(std::shared_ptr<pmt::pmt_base>, unsigned long&)
0000000000032010 T pmt::u64vector_elements(std::shared_ptr<pmt::pmt_base>)
000000000002e560 T pmt::u64vector_elements(std::shared_ptr<pmt::pmt_base>, unsigned long&)
0000000000038170 T pmt::uniform_vector_elements(std::shared_ptr<pmt::pmt_base>, unsigned long&)

```

ncorgan, 2021-05-14:
I'm not sure if these were meant to be aliases or are left over from a refactor, but they're there with no implementation.

willcode, 2021-05-14:
Looks like some sort of cut/paste error during a rename. I think all the pmt_ versions should be removed from pmt.h.

jacobagilbert, 2021-05-14:
Agree with @willcode. Strange this lingered for so long. Good find.

closed 2021-05-19 by mormj