gnuradio/gnuradio #4624: gr-digital: Deprecated CMA Equalizer is still present in MPSK Example

solomonbstoner, 2021-05-16:
In commit 65bd288a2804ef10c5ccac322cb5209907bb5d96, the `digital_cma_equalizer_cc` block was removed. The MPSK example in `gr-digital/examples/mpsk_stage6.grc` needs to be updated to a different block. It appears to be the only file that still uses the CMA equalizer (checked by grepping the id across the gnuradio root folder). 

```
$ grep -ri "digital_cma_equalizer_cc" *
gr-digital/examples/mpsk_stage6.grc:- name: digital_cma_equalizer_cc_0
gr-digital/examples/mpsk_stage6.grc:  id: digital_cma_equalizer_cc
gr-digital/examples/mpsk_stage6.grc:- [digital_cma_equalizer_cc_0, '0', digital_costas_loop_cc_0, '0']
gr-digital/examples/mpsk_stage6.grc:- [digital_pfb_clock_sync_xxx_0, '0', digital_cma_equalizer_cc_0, '0']
```

I would volunteer to offer a fix, but I have no idea how different equalizers work (thats why Im looking at the example).

solomonbstoner, 2021-05-16:
Gimme a while. Im gonna try to propose a change

duggabe, 2021-05-16:
that is the flowgraph used in https://wiki.gnuradio.org/index.php/Guided_Tutorial_PSK_Demodulation#Equalizers
we can work together on this if you like.

solomonbstoner, 2021-05-16:
@duggabe thatll be awesome! Are we going to amend both the example grc and the documentation? I see the LMS DD Equalizer was also deprecated and removed from grc as well

duggabe, 2021-05-16:
yes. the tutorial was the reason I stored the GRC file in the first place. we need to find a suitable replacement, make flowgraphs to show it, and then do a PR on the final one in gr-digital.

solomonbstoner, 2021-05-16:
Ok. I will close the PR that I currently have. How would we coordinate and split the work?

duggabe, 2021-05-17:
Let's start with identifying a good replacement block and see if we can make the flowgraph work with it. then we'll have to go back to recreate some of the intermediate steps in the tutorial.

solomonbstoner, 2021-05-17:
I did that in my PR. ill reopen it as a draft PR. Shall we continue the discussion there?

solomonbstoner, 2021-05-20:
PR merged. Closing issue

closed 2021-05-20 by solomonbstoner

solomonbstoner, 2021-05-23:
Barry has amended the wiki to reflect the changes. Issue resolved