gnuradio/gnuradio #4627: Installed GRC .desktop file doesn't set PYTHONPATH, LD_LIBRARIES_PATH if installed in prefix

marcusmueller, 2021-05-16:
@solomonbstoner found this in his GNU Radio source installation (which put a symbol on his desktop, which doesn't work):

If I do 

```shell
cmake -DCMAKE_INSTALL_PREFIX=/mithlond ..
make -j
make install
```

then the files installed to `/mithlond/share/gnuradio/grc/freedesktop/gnuradio-grc.desktop` and `/mithlond/share/applications/gnuradio-grc.desktop` contain
```desktop
Exec=gnuradio-companion %F
```
which doesn't work.

# Option 1 

They instead should contain something like (line breaks should be removed, just for readability):
```desktop
Exec=env PYTHONPATH=/mithlond/wherever/python/modules/get/installed
         LD_LIBRARY_PATH=/mithlond/wherever/shared/libraries/get/installed:/mithlond/wherever/python/modules/get/installed
         /mithlond/full/path/to/gnuradio-companion %F
```
The problem with the latter would be that it *replaces* these environment variables, instead of amending them. It would solve a common problem with minimal intrusiveness, but it might be an unwelcome change for some users.

# Option 2

The other option is to call

```desktop
Exec=/mithrond/full/path/to/gnuradio_wrapper gnuradio-companion
```

which really only contains something like

```shell
#!/bin/sh
# can't use var expansion, this isn't bash/zsh/... any other fancy shell, as it needs to be portable :(

GRPREFIX=/mithrond
BINDIR=bin/or/whatever/thefolder/was/that/CMake/decided/to/install/executables/into
LIBDIR=lib/or/whatever/thefolder/was/that/CMake/decided/to/install/libraries/into
PYTHONDIR=lib/pythonVERSION.VERSION/site-libs/or/whatever/cmake/determined

export PATH=$GRPREFIX/$BINDIR:$PATH

if [ -z "$PYTHONPATH" ] ; then
  export PYTHONPATH=$GRPREFIX/$LIBDIR:$GRPREFIX/PYTHONDIR
else
  export PYTHONPATH=$PYTHONPATH:$GRPREFIX/$LIBDIR:$GRPREFIX/PYTHONDIR
fi

if [ -z "$LD_LIBRARY_PATH" ] ; then
  export LD_LIBRARY_PATH=$GRPREFIX/$LIBDIR:$GRPREFIX/PYTHONDIR
else
  export LD_LIBRARY_PATH=$GRPREFIX/$LIBDIR:$GRPREFIX/PYTHONDIR:$LD_LIBRARY_PATH
fi

if [ -z "$GRC_BLOCKS" ] ; then
# you get the idea
…
else
…
fi

# finally, start the specified program with specified arguments
# since this does nothing if nothing was specified, we can also 
# source /path/to/this/script
# to set up the GNU Radio environment
$*
```

The appropriate variables should be inserted by CMake into a `gnuradio-grc.desktop.in` or `gnuradio_wrapper.sh.in` file.

# Discussion

While Option 1 seems less invasive, Opt. 2 seems superior, but is a whole new source of strangenesses we might encounter and thus might initially lead to higher maintenance workload.

Wonder how other projects are doing this.

solomonbstoner, 2021-05-16:
Thanks for opening the issue Marcus. Just to clarify, the installation didnt put the `.desktop` icon on my desktop. Is the installer supposed to install a desktop icon by default? I manually copied `./grc/scripts/freedesktop/gnuradio-grc.desktop` over to my desktop `~/Desktop/gnuradio-grc.desktop`. Running it with its default options gave me the ModuleNotFoundError error just as you describe, and I fixed it with option 1. 

I face the same issue with the gnuradio launcher from the Ubuntu search bar. Exact same problem, which I solved by adding the export commands to `~/.profile` or `/etc/profile.d/`.  Not sure if that icon is the same as the `.desktop` icon

[image]

Also, about Option 1. Forgive my ignorance, but is the intrusiveness temporary or permanent? I mean - when run, does it only amend the environment variables for the period of the run, and reverts back to its original values before the run? Or does the change last until explicitly changed to a different value? I did a simple test. It appears to be the former. Why would it be bad if its only temporary?

solomonbstoner, 2021-05-17:
My 2 cents.

##### Option 1

1. It doesnt seem conventional for `.desktop` to specify environment variables. I looked through a couple of `.desktop` files on Github. None I saw use `exec=/usr/bin/env ...`. All just include the executable name, and some specify arguments. But none run  `/usr/bin/env`.

2. I may have tested it wrong, but Option 1 does not appear to modify the environment variables with permanent effect. After running and quitting gnuradio from the desktop icon, the terminal appears to show the same unreplaced environment variable.

In my test I initiated an environment variable in `~/.profile` with a simple "testString". 
```
$ echo $TEST
testString
```
After running `gnuradio-grc.desktop` icon with `Exec=/usr/bin/env TEST=/usr/local/lib gnuradio-companion %F`, I went back to the same terminal. The environment variable TEST still returned "testString". 

##### Option 2

1. I couldnt find another project that has an environment wrapper script like that for their programs. They either have the user modify their environment variable with `export` commands, or modify it automatically through a script (e.g. Anaconda)

[image]

We could modify the installation script to add the environment var modification into `~/.profile` (see Option 3) the same way Anaconda modified my `~/.zshrc` for me.

##### Option 3

I could propose an Option 3. Its not so much a fix as it just involves telling the user to include the `export` commands in [Step E of the wiki](https://wiki.gnuradio.org/index.php/ModuleNotFoundError) in either `~/.profile` or a new script in `/etc/profile.d`. I have already taken the liberty to modify the wiki. I tested both and both fixed my desktop icon launch problem. I believe `~/.profile` modifies the environment variable for a specific user, unlike `/etc/profile.d`, so it should address the concern for computers with multiple users.

marcusmueller, 2021-06-04:
I'd say your Option 3 doesn't really solve the issue that "an icon to click on should just work after installation", so as much as that seems elegant, I don't think that's what we're looking for. 

Option 2:  we certainly can't modify a user's ~/.profile; that's a bit of a breach of trust (and counter-productive, as they might be *intentionally* installing it to a separate prefix that they don't want to "pollute" the rest of their system).

Option 1: yes, it only sets the environment variables for the process started that way – that's the idea!

jaredd, 2021-07-21:
I'm surprised option 2 from @marcusmueller OP using a wrapper or really a startup script is not preferred. There are many examples of these kinds of utilities in Linux presently. Gnuradio-companion is itself a Python script. I would think that that a shells script would be the preferred method.

One point I'd like to make is that there are OOT modules that do different things with their installation. For example, my current source install from maint-3.8 puts GR Python modules in <prefix>/lib/python3/dist-packages. But UHD puts some of its packages in <prefix>/lib/python3.9/site-packages. TBF those seem part of its native python interface as opposed to gr-uhd.
To be safe I currently have a script that basically scans <prefix> for libXX/pythonYYY directories to add to PYTHONPATH. 

This is all to say that it might not be enough to use only those paths from the base GR installation if all OOT packages are to be supported. Of course I haven't tried all OOTs either.