gnuradio/gnuradio #4628: grc: enable debug log

dl1ksv, 2021-05-17:
grc has the comman line option --log, which can be set to debug. Doing so enables the debug log in main.py but not in the started Application.py.

For debugging the output of Application.py would be more important.

How can I enable this log on the command line?

mpentler, 2021-10-17:
I have found that changing __name__ to 'grc' in the logging enabler in each of the source files makes it work fine. Is this all that is needed? Does it need doing in *every* source file? I am happy to take this on, perhaps.

I was about to move a logging statement gui/Application.py to gui/MainWindow.py (where it makes more logical sense to sit) and found that it wasn't actually logging anything to screen until I replicated what is in main.py

dkozel, 2021-10-18:
This comment suggests that loggers need to be formatted as `getLogger('grc.'+__name__)`. 
https://github.com/gnuradio/gnuradio/blob/48120c39228ead7941140be310580188c29b61c1/grc/main.py#L40-L51

But the names of the loggers were set to `__name__` in the same commit as the logging was added/refactored.
https://github.com/gnuradio/gnuradio/commit/03af90b98904b636b6fca75c3a4837391e3c03e2

@sdh11 can you shed any light here? Are we missing some bit of info to use the GRC debug logging correctly?

marcusmueller, 2024-01-28:
I believe this would be an important enhancement to gtk-grc. But we're switching horses to Qt-GRC, so this is something we'll not realistically be able to improve in the near or medium future

closed 2024-01-28 by marcusmueller

dl1ksv, 2024-01-29:
The same problem exists in QT-GRC