gnuradio/gnuradio #4630: Cmake error : Target thrift was not found

NicolasCaseau, 2021-05-17:
Hello, 

I'm currently following the tutorial on the GNU Radio Wiki on how to build an OOT module. I'm now trying to use the cmake command (I'm working on a conda environnement) to build my module but I get the following error  : 

```
(gnuradio) [Nicolas@ppsle-asx-n build]$ cmake -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX -DCMAKE_PREFIX_PATH=$CONDA_PREFIX ../
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/Nicolas/anaconda3/envs/gnuradio/bin/x86_64-conda-linux-gnu-c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/Nicolas/anaconda3/envs/gnuradio/bin/x86_64-conda-linux-gnu-cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.27.1") 
-- Found LOG4CPP: /home/Nicolas/anaconda3/envs/gnuradio/lib/liblog4cpp.so
-- Checking for module 'gmp'
--   Found gmp, version 6.2.1
-- Found GMP: /home/Nicolas/anaconda3/envs/gnuradio/lib/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Found MPIR: /home/Nicolas/anaconda3/envs/gnuradio/lib/libmpirxx.so  
-- Found MPLIB: /home/Nicolas/anaconda3/envs/gnuradio/lib/libgmpxx.so  
-- Found Boost: /home/Nicolas/anaconda3/envs/gnuradio/lib/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.74.0") found components: date_time program_options filesystem system regex thread 
-- Found Volk: Volk::volk  
-- User set python executable /home/Nicolas/anaconda3/envs/gnuradio/bin/python
-- Found PythonInterp: /home/Nicolas/anaconda3/envs/gnuradio/bin/python (found version "3.9.4") 
-- Found PythonLibs: /home/Nicolas/anaconda3/envs/gnuradio/lib/libpython3.9.so (found suitable exact version "3.9.4") 
-- Found PythonLibs: /home/Nicolas/anaconda3/envs/gnuradio/lib/libpython3.9.so
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Found pybind11: /home/Nicolas/anaconda3/envs/gnuradio/include (found version "2.6.2" )
-- Checking for module 'thrift'
--   Found thrift, version 0.14.1
-- Binary 'thrift' not found.
-- Found Git: /usr/bin/git  
-- Found Doxygen: /home/Nicolas/anaconda3/envs/gnuradio/bin/doxygen (found version "1.9.1 (ab553a81f1725c385ca98109c269b0222389b72a*)") found components: doxygen missing components: dot
-- Using install prefix: /home/Nicolas/anaconda3/envs/gnuradio
-- Building for version: v1.0-compat-xxx-xunknown / 1.0.0git
-- No C++ unit tests... skipping
-- PYTHON and GRC components are enabled
-- Python checking for pygccxml - not found
-- Configuring done
CMake Error at /home/Nicolas/anaconda3/envs/gnuradio/share/cmake/pybind11/pybind11Tools.cmake:166 (add_library):
  Target "howto_python" links to target "Thrift::thrift" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /home/Nicolas/anaconda3/envs/gnuradio/lib/cmake/gnuradio/GrPybind.cmake:239 (pybind11_add_module)
  python/bindings/CMakeLists.txt:34 (GR_PYBIND_MAKE_OOT)

CMake Error at lib/CMakeLists.txt:24 (add_library):
  Target "gnuradio-howto" links to target "Thrift::thrift" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

As you can see it seems that the target "Thrift" could not be found. Could you tell me if you have any intel on how I could fix this issue ? Thanks a lot 

Nicolas

solomonbstoner, 2021-05-17:
Im not sure how Conda works, but `sudo apt install thrift-compiler` gave me the `thrift` binary needed for my regular build

NicolasCaseau, 2021-05-18:
I installed thift-compiler and it works. Thank you very much for your answer !

solomonbstoner, 2021-05-18:
Youre most welcome. If theres no further issues, you may close the issue :)

closed 2021-05-18 by NicolasCaseau

ryanvolz, 2021-05-18:
Hey, glad you two worked this out between yourselves! You may have noticed that I've been pushing a lot of `gnuradio` conda package updates over the past week, and this was a case of one of those changes (adding thrift support) causing undesirable downstream effects. I've already made #4618 to address this particular problem upstream (OOTs shouldn't actually need thrift to compile), and the latest `gnuradio` conda package has this fixed so that you shouldn't need to install `thrift-compiler` anymore for OOTs. But it shouldn't hurt to have it anyway!