gnuradio/gnuradio #4631: Gnuradio 3.8 AttributeError: 'top_block_sptr' object has no attribute 'tabid_grid_layout_0'

Letiel-88, 2021-05-17:
In this simple flowgraph:
[image]
[image]
[image]

I'm getting the following error:
Traceback (most recent call last):
  File "/home/aaa/Desktop/Prova.py", line 196, in <module>
    main()
  File "/home/xxx/Desktop/Prova.py", line 172, in main
    tb = top_block_cls()
  File "/home/aaa/Desktop/Prova.py", line 128, in __init__
    self.Tab_grid_layout_0.addWidget(self._qtgui_time_sink_x_0_win, 1, 1, 1, 1)
  File "/home/aaa/sdr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 93, in __getattr__
    return getattr(self._impl, name)
AttributeError: 'top_block_sptr' object has no attribute 'Tab_grid_layout_0'

>>> Done (return code 1)

Gnuradio Companion Version: v3.8.2.0-118-g8b2012ab (Python 3.6.9)

ChristopheSeguinot, 2021-05-17:
This is a bug currently under investigation ( see Pull Request #4629). 
A simple temporary workaround is to add a GUI Hint in every QT GUI Tab widget for example `0,0` in yours

dl1ksv, 2021-05-19:
This error happens on the master branch, too.

willcode, 2021-06-02:
Related PRs merged. Please re-open if there is still a problem.

closed 2021-06-02 by willcode