gnuradio/gnuradio #4634: Interleave block: Unable to modify Vector Length

solomonbstoner, 2021-05-18:
I was just playing around with vectors to learn something about it. I tried to change the interleaver block from Vector Length of 3 to Vector Length of 1. I was not allowed to directly change the Vector Lengths for all the blocks from 3 to 1. Doing so gave me the error below for the bottom vector source:

> Source IO size "12" does not match sink IO size "4".

[image]

Instead, to make the change, I had to go the long route. So, back to beginning, everything is fine with Vector Length of 3.

[image]

I change Vector Length of 3 to 1 in the Interleave block only. The others are not changed yet. Only the upper input and output ports have changed from vector length of 3 to 1, shown in the red arrow indicating an error. The bottom input port somehow still has a vector length of 3.

[image]

I change the Vector Length only for the blocks with the errors. Now there are _no more errors_ . Note that the vector source at the bottom still have a Vector Length of 3. All other blocks have their Vector Length changed from 3 to 1. I can generate the flowgraph and run, but _nothing gets shown_. The QT Time sink does not show.

[image]

If I were to manaully change the Vector Length of the bottom vector source myself, an error appears. I cannot make that change.

[image]

The only way I could was to double click on the Interleave block to enter its properties, then click "Cancel". Clicking on "Ok" does not work. Only then does the error go away and the program runs properly (ie QT Time sink appears).

solomonbstoner, 2021-05-18:
Imma try to fix this issue myself

willcode, 2021-05-18:
There's something very strange going on in the GRC connection validator code. I tried out this example and a couple similar ones and couldn't make sense of what it was doing.

solomonbstoner, 2021-05-19:
Interesting. It appears like the Interleave block is not the only block. Even the Add and Subtract blocks have the same issue. 

And also tested with 3 and 4 inputs. Same issue. Only the top input and the output port gets updated while every other port is left unchanged

[image]

commit 6f580396 (2021-05-24): grc: Fix vector length bug

Make all ports follow the `vlen` of their master port the same way it is
for `dtype`. Fixes #4634.
PR #4654 (2021-05-24): grc: Fix vector length bug (backport to maint-3.8)
Make all ports follow the `vlen` of their master port the same way it is
for `dtype`. Fixes #4634.

(cherry picked from commit 6f5803963c56d454564ba2ee2e5f2607689b3c43)

Backport https://github.com/gnuradio/gnuradio/pull/4637
commit cba012fb (2021-05-24): grc: Fix vector length bug

Make all ports follow the `vlen` of their master port the same way it is
for `dtype`. Fixes #4634.

(cherry picked from commit 6f5803963c56d454564ba2ee2e5f2607689b3c43)
PR #4655 (2021-05-24): grc: Fix vector length bug (backport to maint-3.9)
Make all ports follow the `vlen` of their master port the same way it is
for `dtype`. Fixes #4634.

(cherry picked from commit 6f5803963c56d454564ba2ee2e5f2607689b3c43)

Backport https://github.com/gnuradio/gnuradio/pull/4637
closed 2021-05-24 by mormj