gnuradio/gnuradio #4635: Cross compiling issue: Could NOT find LOG4CPP library

cafeclimber, 2021-05-18:
I am following [this guide](https://wiki.gnuradio.org/index.php/Cross_compile_for_Raspberry_Pi) to cross compile for raspberry pi. I have been able to get all the way to the cmake step, but run into the following

<details>
    <summary>Terminal Output</summary>

```
$ cmake -DCMAKE_INSTALL_PREFIX=/home/$USER/usr/local/ -DENABLE_DOXYGEN=OFF -DENABLE_SPHINX=OFF -DENABLE_GR_DTV=OFF -DENABLE_GR_FEC=OFF -DENABLE_GR_CHANNELS=OFF -DENABLE_GR_AUDIO=OFF -DENABLE_GR_TRELLIS=OFF -DENABLE_GR_VIDEO_SDL=OFF -DENABLE_GR_VOCODER=OFF -DENABLE_GR_WAVELET=OFF -DENABLE_GR_ZEROMQ=off -DCMAKE_TOOLCHAIN_FILE=../cmake/Toolchains/aarch64-linux-gnu.cmake ..
-- Build type not specified: defaulting to release.
-- Build type set to Release.
--
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
--
-- Configuring testing-support support...
--   Enabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
-- Extracting version information from git describe...
-- Compiler Version: aarch64-linux-gnu-gcc (Ubuntu 10.2.0-8ubuntu1) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/aarch64-linux-gnu-gcc:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/aarch64-linux-gnu-g++:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- Could NOT find Boost (missing: date_time program_options filesystem system regex thread unit_test_framework) (found suitable version "1.67.0", minimum required is "1.53")
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.6", minimum required is "2.7.6")
-- Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS) (Required is exact version "3.8")
-- Python checking for six - python 2 and 3 compatibility library - found
--
-- Checking for module SWIG
-- Found SWIG version 4.0.1.
--
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = FALSE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Dependency SIX_FOUND = TRUE
--   Disabling python-support support.
--   Override with -DENABLE_PYTHON=ON/OFF
--
-- Configuring VOLK support...
-- Build type set to Release.
-- Extracting version information from git describe...
-- Building Volk without cpu_features
--
-- Python checking for distutils
-- Python checking for distutils - found
--
-- Python checking for python >= 3.4
-- Python checking for python >= 3.4 - found
--
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - found
-- Checking for module 'orc-0.4 > 0.4.11'
--   No package 'orc-0.4' found
-- orc files (missing: ORC_LIBRARY)
-- QA Testing is enabled.
--   Modify using: -DENABLE_TESTING=ON/OFF
-- System profiling is disabled.
--   Modify using: -DENABLE_PROFILING=ON/OFF
-- Compiler name: GNU
-- Architecture is not x86 or x86_64, Overruled arch 3dnow
-- Architecture is not x86 or x86_64, Overruled arch mmx
-- Architecture is not x86 or x86_64, Overruled arch sse
-- Architecture is not x86 or x86_64, Overruled arch sse2
-- Architecture is not x86 or x86_64, Overruled arch sse3
-- Architecture is not x86 or x86_64, Overruled arch ssse3
-- Architecture is not x86 or x86_64, Overruled arch sse4_a
-- Architecture is not x86 or x86_64, Overruled arch sse4_1
[... 630 lines cut ...]
# Create a config.h with some definitions to export to other projects.
CONFIGURE_FILE(
  ${CMAKE_CURRENT_SOURCE_DIR}/config.h.in
  ${CMAKE_CURRENT_BINARY_DIR}/gnuradio-runtime/include/gnuradio/config.h
)

#Re-generate the constants file, now that we actually know which components will be enabled.
configure_file(
    ${CMAKE_CURRENT_SOURCE_DIR}/gnuradio-runtime/lib/constants.cc.in
    ${CMAKE_CURRENT_BINARY_DIR}/gnuradio-runtime/lib/constants.cc
    ESCAPE_QUOTES
@ONLY)

# Install config.h in include/gnuradio
install(
    FILES
    ${CMAKE_CURRENT_BINARY_DIR}/gnuradio-runtime/include/gnuradio/config.h
    DESTINATION ${GR_INCLUDE_DIR}/gnuradio
)
```
</details>

I also made sure to `git checkout maint-3.8` and run `git submodule update --init` which the guide did not say to do, but I assumed was a correct step.

Let me know if there's any more information needed.

marcusmueller, 2021-05-28:
hey, sorry for the long silence.

Indeed, finding log4cpp has proven to be a problem, even in self-hosted (i.e. non-cross) builds.

I kind of hate to recommend bad hotfixes, but does explicitly specifying [`-DLOG4CPP_INCLUDE_DIR`](https://github.com/gnuradio/gnuradio/blob/master/cmake/Modules/FindLOG4CPP.cmake#L4) and [`-DLOG4CPP_LIBRARIES`](https://github.com/gnuradio/gnuradio/blob/master/cmake/Modules/FindLOG4CPP.cmake#L5) change the situation?

ryanvolz, 2021-05-28:
I think specifying `-DCMAKE_PREFIX_PATH=/home/$USER/usr/local/` with the `cmake` command is also likely to help it find `log4cpp`, assuming you also have `log4cpp` installed to that prefix. If it's somewhere else, set that path as the `CMAKE_PREFIX_PATH` instead.

NebulaFinder, 2021-07-13:
> I think specifying `-DCMAKE_PREFIX_PATH=/home/$USER/usr/local/` with the `cmake` command is also likely to help it find `log4cpp`, assuming you also have `log4cpp` installed to that prefix. If it's somewhere else, set that path as the `CMAKE_PREFIX_PATH` instead.

hello ryanvolz, I try to use cmake -DCMAKE_PERFIX_PATH to solve this problem, but it seem to be useless.

mormj, 2022-01-20:
Closing as this won't be fixed since we have abandoned Log4CPP for spdlog in GR 3.10

closed 2022-01-20 by mormj