gnuradio/gnuradio #4639: Incorrect prefix in package installed on Ubuntu 20.04.2 from ppa:gnuradio/gnuradio-releases

bmarjanovic, 2021-05-20:
I installed gnuradio 3.9. on Ubuntu 20.04.2 using ppa:gnuradio/gnuradio-releases with no issues. The companion seems to work fine. However, I can not create new module using modtool 'out of the box'.
```
$ gr_modtool newmod temp
ModToolException: Could not find gr-newmod source dir.
```
Interestingly, running config info returns prefix '/.'
```
$ gnuradio-config-info --prefix
/.
```
This would explain why gr.prefix() which is imported via 'form gnuradio import gr' returns the string that creates an incorrect path to search for templates and later to find clang-format.conf file. Having the correct path in /etc/gnuradio/conf.d/modtool.conf makes no difference. I copied the file to ~/.gnuradio/config.conf  but still did not work. For some reason it seems these files are not checked for the path.

The code inspection shows there is new 'srcdir' parameter for a newmod option . If srcdir is provided, the module seems to be created, 
```
$ gr_modtool newmod --srcdir /usr/share/gnuradio/modtool/templates/gr-newmod temp
Creating out-of-tree module in ./gr-temp...
Failed to copy .clang-format: [Errno 2] No such file or directory: '/./share/gnuradio/clang-format.conf'
Done.
Use 'gr_modtool add' to add a new block to this currently empty module.

```
Note that the srcdir provided in the command is the same as the value in the config file. There is still a failure because file clang-format.conf can not be found. The reason is the same as before - gr.prefix() returns incorrect value. But it seems to be 'warning level' since the module scaffolding seems to be in place and blocks can be added. 

The workaround (would not call it a fix) for new module creation is to modify lines creating the path in newmod.py files (under both cli and core directory) in your gnuradio python installation

beroset, 2021-06-05:
I encountered this today installing 3.10 from github head on Fedora 34.  My workaround was to use `export GR_PREFIX=/usr` (which also then lets it find clang-format).

ryanvolz, 2021-06-05:
So that's two different distributions now, both of which install to the `/usr` prefix. I think there must be something funky going on with that prefix path that has to do with it's structure being a special case compared to any other prefix. But I haven't found yet where it goes wrong...

beroset, 2021-06-07:
I have done a little troubleshooting.  Under Fedora, the library `libgnuradio-runtime.so` file is installed (correctly) in `/lib64`.  So the logic in the `prefix()` function in `gnuradio-runtime/lib/constants.cc.in` says to go one directory up from that (`@prefix_relative_to_libs@` evaluates to '../') which results in `/`.  By contrast, under Ubuntu (and probably Debian, but I haven't checked that) the library is installed in `/usr/lib/x86_64-linux-gnu/`.  

Unfortunately, the `prefix` documentation string is a cut-and-paste duplicate of that for `sysconfdir` and so it's obviously wrong and I don't really know what `prefix()` is actually intended to point to under all circumstances and on all platforms.  An obvious "fix", at least for Linux, would be to simply rely on the config step to use `CMAKE_INSTALL_PREFIX` and we'd be done.  I'm guessing that the reason it's calculated dynamically instead is to allow running of gnuradio without installing it.  

Because it's part of the API, I think we need to have a clear definition of what it's supposed to do before tinkering.  Maybe others know that definition?

ryanvolz, 2021-06-07:
Let me describe what I think is happening on Fedora then:

1. `CMAKE_INSTALL_PREFIX` is set to `/usr`
2. `prefix_relative_to_lib` is calculated as the relative path from `/usr/lib64` to `/usr`, which results in `../`.
3. CMake doesn't put the libs into `/usr/lib64` as the above calculation expects, because the `/usr/` prefix is a special case. Instead, the libs go to `/lib64`.
4. Anything that later calls `prefix()` expecting `/usr` fails when it actually returns `/lib64/../` or equivalently `/.`

This is an issue now and not with earlier versions because of #4127, which switched to calculating the prefix using the relative path from the library location so that installations could be relocatable (e.g. for conda).

What should probably happen is a switch to using the CMake `GNUInstallDirs` module, which knows about these special cases. Then we can fix the `prefix_relative_to_lib` calculation so that it gives the correct value of `../usr` in this case.

beroset, 2021-06-08:
A little more investigation reveals that under Fedora, `/lib64` is a symlink to `/usr/lib64`.  Would attempting to expand symlinks be a solution or is it merely a shortcut on the road to insanity?

ryanvolz, 2021-06-08:
> A little more investigation reveals that under Fedora, `/lib64` is a symlink to `/usr/lib64`. Would attempting to expand symlinks be a solution or is it merely a shortcut on the road to insanity?

Well that's a wrinkle. So it is actually installed into the `/usr` prefix, but I guess the library is found from `/lib64` and reports that location instead. Maybe changing the CMake won't have the desired effect then, although it might fix related problems and would be a good improvement anyway. Resolving the symlink when the library reports its location is probably what is necessary, assuming it's possible.

beroset, 2021-06-08:
I will experiment with that approach and maybe even submit a pull request if it looks reasonable to me.

closed 2021-06-16 by mormj