gnuradio/gnuradio #4640: UHD tuner_request_t::policy_t has no Python binding

Codey-McCodeface, 2021-05-20:
I am unable to set the `rf_freq_policy` or `dsp_freq_policy` of a uhd tune request from Python.

Adding the binding for the enum did solve this problem. The changes below worked for me:

```diff
diff --git a/gr-uhd/python/uhd/bindings/uhd_types_python.cc b/gr-uhd/python/uhd/bindings/uhd_types_python.cc
index c593ce480..f05df299c 100644
--- a/gr-uhd/python/uhd/bindings/uhd_types_python.cc
+++ b/gr-uhd/python/uhd/bindings/uhd_types_python.cc
@@ -141,7 +141,8 @@ void bind_uhd_types(py::module& m)
 
     // ::uhd::tune_request_t
     using tune_request_t = ::uhd::tune_request_t;
-    py::class_<tune_request_t>(m, "tune_request_t")
+    py::class_<tune_request_t> tune_request(m, "tune_request_t");
+    tune_request
         .def(py::init<double, double>())
         .def(py::init<double>(), py::arg("target_freq") = 0.0)
         .def_readwrite("target_freq", &tune_request_t::target_freq)
@@ -151,6 +152,13 @@ void bind_uhd_types(py::module& m)
         .def_readwrite("dsp_freq", &tune_request_t::dsp_freq)
         .def_readwrite("args", &tune_request_t::args);
 
+    // ::uhd::tune_request_t::policy_t
+    using policy_t = ::uhd::tune_request_t::policy_t;
+    py::enum_<policy_t>(tune_request, "policy_t")
+        .value("POLICY_NONE", policy_t::POLICY_NONE)
+        .value("POLICY_AUTO", policy_t::POLICY_AUTO)
+        .value("POLICY_MANUAL", policy_t::POLICY_MANUAL)
+        .export_values();
 
     // ::uhd::sensor_value_t
     using sensor_value_t = uhd::sensor_value_t;
```

mormj, 2021-05-20:
@Codey-McCodeface - this appears to the be correct solution to the issue you are seeing - would you mind submitting this as a pull request and reference this issue?

willcode, 2021-06-02:
Cross ref did not trigger auto-close. Manu-closing.

closed 2021-06-02 by willcode