gnuradio/gnuradio #4644: make test qa_tag_share and qa_skiphead have Failed

elmaghba, 2021-05-22:
99% tests passed, 2 tests failed out of 346

Total Test time (real) = 283.26 sec

The following tests FAILED:
	190 - qa_skiphead (Failed)
	198 - qa_tag_share (Failed)
Errors while running CTest
Makefile:129: recipe for target 'test' failed
make: *** [test] Error 8

When I tried ctest -V -R qa_skiphead, I got :
190: Test command: /bin/sh "/home/abdurrahman/workarea/gnuradio/build/gr-blocks/python/blocks/qa_skiphead_test.sh"
190: Test timeout computed to be: 9.99988e+06
190: Segmentation fault (core dumped)
1/1 Test #190: qa_skiphead ......................***Failed    0.85 sec

Thank you so much!

willcode, 2021-05-22:
What O/S and O/S version?

If you are looking at https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux, it mentions a very old version of GNU Radio (3.7).

Take a look at https://wiki.gnuradio.org/index.php/InstallingGR.

If you were using a very old version of GNU Radio, we can close this issue, since the test passes on multiple platforms with newer versions.

elmaghba, 2021-05-23:
Yes, I was using the old version as indicated at: https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux

After following the Ubuntu PPA Installation instructions to install the GR, I found that it might have installed another version of UHD as well. Because I was already had UHD 3.14, but when I run the GnuRadio, it prints out an error and shows the path of UHD 3.010. 

Do you have any idea about that?
Thank you so much!

closed 2021-05-23 by elmaghba