gnuradio/gnuradio #4646: Race condition when dynamically changing bandpass / bandreject filters within embedded python block ?

eggelene, 2021-05-22:
I have created a graph that changes band-pass / band-reject filter parameters dynamically within embedded python block ( inside the work function. The filter updates are timed by this python code  ). After updating to 3.9.1.0 I noticed that my output is not always consistent. Meaning in 1 of the 7 runs I see that the filters are not updated correctly in one case. In most cases the output file is what I expect, and in some cases it is not ( no other changes are made to system/code). This makes me believe there is a race condition somewhere. When the error happens, no error or warning is shown in the log window.

**Question1**: Is it supported by GNU radio that I update a variable inside the work function of an embedded python block. This variable is used in 2 other filters. See below. I take care to update the low and high in the right order. If this is not the best way of doing it, can I get a pointer how to do it with methods that are supported. 

Generated code:
```
def set_filter1_low_cut(self, filter1_low_cut):
        self.filter1_low_cut = filter1_low_cut
        Qt.QMetaObject.invokeMethod(self._filter1_low_cut_line_edit, "setText", Qt.Q_ARG("QString", eng_notation.num_to_str(self.filter1_low_cut)))
        self.band_pass_filter_0_0.set_taps(firdes.complex_band_pass(1, self.samp_rate, self.filter1_low_cut, self.filter1_high_cut, self.filter1_transitionwidth, window.WIN_KAISER, 6.76))
        self.band_reject_filter_0_0.set_taps(firdes.complex_band_reject(1, self.samp_rate, self.filter1_low_cut, self.filter1_high_cut, self.filter1_transitionwidth, window.WIN_KAISER, 6.76))
```
**Question2**: how is the synchronization done when I update the filters within the work function ( assuming multiple threads are running) ? Are these filter used when processing the next block ? Or they they immediately used ( update filters while processing a block of samples ). What happens if I update lowCut and highCut after each other ( there is a bit of time in between). Are some samples processed with only updated the lowCut. Does GNU radio guaranty that the filters are updated in graph a the same timestamp even when some paths have more filters ( delay) in the chain ?

**Question3**: I never noticed it before with GNUradio 3.9.0.0 ( not sure if it never happened), could some changes in the code trigger this ?

System: Ubuntu 20.4
Intel Core i7 (hyper threading enabled)
Install: 
`$ sudo add-apt-repository ppa:gnuradio/gnuradio-releases`
volk_profile
 
Problem description:
The variable are correctly updated in the QT view. ( So this should exclude my python script as this done within a single `set_filter1_low_cut call ` see above). Based on the view I looks like the band-pass filters are not updated ( only the high cut of band-pass filter1 is stuck at the old value) while the band-reject filters are correctly updated. Based on these findings the problems seems to come from GNUradio and not from my code.

[image]

[image]

willcode, 2021-05-22:
Calls to work() happen whenever the scheduler see that work can be done, so you should not depend on the order of block work() functions being called.

Code inside work() should operate only on block data (streams, tags, messages, internal state). In Python, nothing keeps you from changing things outside the block, but the behavior would be, as you've found, unpredictable. You could send a message from a block work() call, then use a Message Pair to Var to set a variable (I haven't tried this).

Updates to variables (including taps) take place "as soon as possible". The exact timing and order of updates is not specified.

Closing, since this is not an issue with (our) intended behaviour of GNU Radio. Please feel free to put in an issue for a feature request for some sort of synchronization. This sounds like a good goal, but needs definition, and won't be implemented any time soon. For questions, the mailing list or chat.gnuradio.org would work well.

closed 2021-05-22 by willcode

eggelene, 2021-05-22:
I just tried the solution with **Message Pair to Var**. Messages from the Embedded Python block results into a **None** value. Messages from e.g. QT GUI Toggle Button work. I sent an e-mail to mailing list hoping that someone can help me with this.

[image]

[image]
```
"""
Embedded Python Blocks:

Each time this file is saved, GRC will instantiate the first class it finds
to get ports and parameters of your block. The arguments to __init__  will
be the parameters. All of them are required to have default values!
"""

import pmt
import numpy as np
from gnuradio import gr

class blk(gr.sync_block):  # other base classes are basic_block, decim_block, interp_block
    """Embedded Python Block example - a simple multiply const"""

    def __init__(self, example_param=1.0):  # only default arguments here
        """arguments to this function show up as parameters in GRC"""
        gr.sync_block.__init__(
            self,
            name='Embedded Python Block',   # will show up in GRC
            in_sig=[np.complex64],
            out_sig=[np.complex64]
        )
        self.message_port_register_out(pmt.intern('msg_out'))
        # if an attribute with the same name as a parameter is found,
        # a callback is registered (properties work, too).
        self.example_param = example_param
        self.sendMessage=0
        self.sendMessageCount=123

    def work(self, input_items, output_items):
    
        print (".%d." % (self.sendMessage),end="")
        self.sendMessage=self.sendMessage+len(output_items[0])
        if (self.sendMessage>=32000):
            self.sendMessage=0
            
            key0 = pmt.intern("test")
            val0 = pmt.from_long(self.sendMessageCount)
            
            msg = pmt.make_dict()
            msg = pmt.dict_add(msg, key0, val0)
            
            self.message_port_pub(pmt.intern('msg_out'), msg);
            
            self.sendMessageCount=self.sendMessageCount+1
            
        """example: multiply with constant"""
        output_items[0][:] = input_items[0] * self.example_param
        return len(output_items[0])
```

Generating: '/home/rfstreamer/Documents/GNURadio/message.py'

Executing: /usr/bin/python3 -u /home/rfstreamer/Documents/GNURadio/message.py

```
.0..4096..8191..12287..14335..15359..15871..16127..16255..16319..16351..16367..16375..16379..16381..16382..20478..22526..23550..24062..24318..24446..24510..24542..24558..24566..24570..24572..24573..28669..30717..31741.******* MESSAGE DEBUG PRINT ********
((test . 123))
************************************
.0..256..384..448..480..496..504..508..510..511..4607..6655..7679..8191..8447..8575..8639..8671..8687..8695..8699..8701..8702..12798..14846..15870..16382..16638..16766..16830..16862..16878..16886..16890..16892..16893..20989..23037..24061..24573..24829..24957..25021..25053..25069..25077..25081..25083..25084..29180..31228.******* MESSAGE DEBUG PRINT ********
((test . 124))
************************************
.0..512..768..896..960..992..1008..1016..1020..1022..1023..5119..7167..8191..8703..8959..9087..9151..9183..9199..9207..9211..9213..9214..13310..15358..16382..16894..17150..17278..17342..17374..17390..17398..17402..17404..17405..21501..23549..24573..25085..25341..25469..25533..25565..25581..25589..25593..25595.******* MESSAGE DEBUG PRINT ********
(test . 200)
************************************
.25596..29692..31740.******* MESSAGE DEBUG PRINT ********
((test . 125))
************************************
.0..512..768..896..960..992..1008..1016..1020..1022..1023.******* MESSAGE DEBUG PRINT ********
(test . 100)
************************************
.5119..7167..8191..8703..8959..9087..9151..9183..9199..9207..9211..9213..9214..13310..15358..16382..16894..17150..17278..17342..17374..17390..17398..17402..17404..17405..21501..23549..24573..25085..25341..25469..25533..25565..25581..25589..25593..25595..25596..29692..31740.******* MESSAGE DEBUG PRINT ********
((test . 126))
************************************
.0..512..768..896..960..992..1008..1016..1020..1022..1023.
```