gnuradio/gnuradio #4656: gr-soapy: Custom sink Center Freq fails with one channel selected.

drmpeg, 2021-05-24:
The block requires than the center freq for channel 1 also be set, but it's impossible to do that.

willcode, 2021-05-24:
Which module?

willcode, 2021-05-24:
Can not duplicate. Tried the Soapy Custom Sink with both PlutoSDR and BladeRF, configured like this:

[image]
[image]

and also tested with periodic frequency change messages:

[image]

Any more info on your setup?

drmpeg, 2021-05-25:
Try it without a variable named "freq".

closed 2021-05-25 by willcode