gnuradio/gnuradio #4666: soapy: value_in_range does not check range step

willcode, 2021-05-25:
Where a `range_t` (`SoapySDR::Range`) has a step defined, need to check to see if value is on a step (or approximately?). For example, some soapy drivers have discrete sample rates, defined as one or more min/max/step ranges.

willcode, 2021-05-25:
@ncorgan how do we define "close enough" here? For example, soapyplutosdr sample rate step should be 1, 4, or 8 hz. We don't need to bother the user about this because the clock is going to be off by more than that already. So in this specific case, it can just be ignored.

ncorgan, 2021-06-11:
The simplest solution would be some fixed epsilon, ideally tighter than drivers would likely enforce for their errors. If we're bringing in this uncertainty, I'm wondering if the relevant functions put out a warning instead of throwing if this third step check doesn't pass. A "This value doesn't look right, but we'll try it anyway" type of thing.