gnuradio/gnuradio #4668: Broken Band-reject Filter Taps block

devnulling, 2021-05-26:
The `Window` select does not seem to be parsed correctly in the `Band-reject Filter Taps` block:

```
Value "firdes.band_reject(gain, samp_rate, low_cutoff_freq, high_cutoff_freq, width, win, beta)" cannot be evaluated:
band_reject(): incompatible function arguments. The following argument types are supported:
    1. (gain: float, sampling_freq: float, low_cutoff_freq: float, high_cutoff_freq: float, transition_width: float, window: gnuradio.fft.fft_python.window.win_type = win_type.WIN_HAMMING, param: float = 6.76) -> List[float]

Invoked with: 1.0, 40000000.0, 1000000, 1100000, 1000, None, 6.76

Param - Window(win):
	Expression None is invalid for type 'int'.
```

Also it doesn't seem to be populating the `Real`/`Complex` option param that was added here https://github.com/gnuradio/gnuradio/commit/0f045e515352973a4effe5dc50b7bccf7b1ae371#diff-a58d597700dda72970012f2d238b9f51d0a1e6bc79ec0b335ffae1529873a212

Tested on master 44152d6c3fb19ab56ef6f839f63fbfb26650b46a

devnulling, 2021-05-26:
Looks like this also affects the `Band Pass Filter`, `Band Reject Filter` and `Band-pass Filter Taps` blocks.

drmpeg, 2021-05-26:
Works fine here on latest master. You may want to try deleting ~/.cache/grc_gnuradio/cache.json (old versions of that file can cause really weird issues, especially when switching between 3.8 and 3.9).

marcusmueller, 2021-05-26:
Calling @Yamakaja in on this to cross-check, please.

Yamakaja, 2021-05-26:
> Also it doesn't seem to be populating the Real/Complex option param that was added here 0f045e5#diff-a58d597700dda72970012f2d238b9f51d0a1e6bc79ec0b335ffae1529873a212

That's because the real/complex option isn't a "normal" parameter like the others, but switches between two methods (`firdes.band_reject` and `firdes.complex_band_reject`, which was added in that commit).

It appears that you're specifying `None` as your Window for some reason - could you show us a screenshot of the block settings? Oh, and please make sure that your cache is clean, like drmpeg suggested.

closed 2021-05-26 by devnulling

mormj, 2021-05-26:
https://github.com/gnuradio/gnuradio/issues/2763 strikes again!