gnuradio/gnuradio #4673: SDRplay with GNU Radio in Ubuntu

juriguen, 2021-05-27:
Hi!

I'm trying to make an SDRplay module by Franco Venturi work in GNU Radio but, after installing the latest app version for Ubuntu 
[image]
and following the instructions of his readme file to build the block, I try to run a very simple example and get the error: ModuleNotFoundError: No module named 'sdrplay3'

Any ideas what might I be doing wrong?

Thanks!

willcode, 2021-05-27:
Help requests are via the mailing list or chat.gnuradio.org. Closing.

closed 2021-05-27 by willcode

juriguen, 2021-05-27:
I don't understand why issues related to GNU radio cannot be here