gnuradio/gnuradio #4680: GNU Radio error: QCoreApplication::sendPostedEvents: Cannot send posted events for objects in another thread

juriguen, 2021-05-28:
Hi again

When trying to build a flow in which certain part of the spectrum is swept, I find the GNU Radio companion outputs this line:
```
QCoreApplication::sendPostedEvents: Cannot send posted events for objects in another thread
```

Thanks!

The program runs, but this message does not disappear. Any ideas why is this happening and what is causing it?

[image]

solomonbstoner, 2021-06-05:
Ah. I had the same issue as well. Are you using a function probe to change the content of a variable that the Freq sink uses on the fly? Cause I did that. Im not so sure what is the cause, but it appears to be about Qt's signals and slots across threads. I havent looked at the implementation of the Freq Sink yet, so I cant say for sure. Perhaps someone else has and can advise further

Jeija, 2022-01-01:
I am running into a similar issue, I also get the following error message and, even worse, my GNURadio flow graph crashes after a few seconds:
```
QCoreApplication::sendPostedEvents: Cannot send posted events for objects in another thread
```

In my application, I am receiving some signal with some arbitrary sample rate (from real hardware, but, in this example, I used a signal source) and probing its rate. My goal is to update the time axis of a QT GUI Time Sink using the probed sample rate.
[image]

You can get the GRC file from here: https://gist.github.com/Jeija/4230595672a3f4367827a34da2e4fa86
With the latest GNURadio build (from git master) and with GNURadio 3.9.4.0, this flow graph will produce aforementioned warning message several times and then crash after a few seconds.

(I know that for the "QT GUI Frequency Sink", I can just set "Show Msg Ports" to "True" and then feed the probed sample rate value into the block as a message, but this does not work with the "QT GUI Time Sink", since it does not have that option.)

Jeija, 2022-01-01:
### Cause of the Bug
What I have figured out is that, in my flow graph, this issue occurs because - apparently - the callback for the sample rate variable is **called from a thread that is different to the QT GUI thread**. The QT GUI blocks are not capable of handling parameter updates from any thread other than the QT main thread.

So, for example, if the sample rate is changed using a GUI slider then that is fine for the "QT GUI Time Sink" block (since the slider also runs in the Qt main thread), but if it is determined using a "Probe Rate" block, then that will potentially crash the application. Does anyone know how we are supposed to deal with this situation? Shouldn't all callback functions of QT GUI blocks defer updates to the Qt main thread via signal and slot?

### Proof and Workaround
To show that this is actually the problem and as a hacky workaround, I have written a special block that takes an input parameter X which can be updated from any thread and updates an
 output parameter Y based on X so that Y = X. However, it defers the update callback for the output parameter Y to the QT GUI main thread using signal and slot: [gr-qtvarupdate.zip](https://github.com/gnuradio/gnuradio/files/7798289/gr-qtvarupdate.zip) (these are the important files: https://gist.github.com/Jeija/8940244b1714d0fd4d6662626aab35d9)
[image]
You can get the GRC file from here: https://gist.github.com/Jeija/78e5dea399c167cefc3e379e5c95ac07

With this workaround block included, the flow graph no longer crashes.

### Questions
So, at this point, my questions would be:

* For setting parameters of Qt blocks from threads other than the Qt main thread, is there already some solution implemented by GNURadio that I just have not come across yet?
* If not, is the workaround block the way to go? Or should all Qt blocks themselves defer GUI updates via callback functions to the main thread using a signal / slot approach? Or should I just never update Qt block parameters from outside the Qt main thread?

delhatch, 2023-09-07:
Thanks Jeija!  Here was my (similar) problem, and my fix:

GOAL:
To change the tuning frequency (variable name: cent_freq) of my SDR, via Python. I used the "GNURadio XMLRPC Server Automation using Standalone Python Code" example.

The variable "cent_freq" was used by the RTL-SDR block, and also two "QT GUI Waterfall Sink" blocks.

SYMPTOM:
The GNU Radio code worked normally, and after launching the Python script (the client) that changed "cent_freq" via XMLRPC, it would work for a few messages (controlling everything perfectly), but then crash.
I got the "Cannot send posted events for objects in another thread" messages, before the crashes.
I noticed that if I disabled the two Waterfall Sinks (but leaving the RTL-SDR block), the SDR would still tune perfectly, and the messages and crashes stopped occurring.
How to restore the Waterfalls?

SOLUTION:
I added a Core->Message Tools->"Variable to Message" block. I set it's Variable to "cent_freq". This is the same variable that the SDR block consumed, and was the variable being changed via XMLRPC.
On the Waterfall blocks, I enabled "Show Msg Ports" and set the "Center Frequency (Hz)" to some default value.
I then linked the "msgout" port to the "freq" port of each Waterfall block.

LIMITATIONS:
This solution only works if the QT blocks that are consuming the variable have a message port that exposes in interface for the variable you need to change.

marcusmueller, 2024-01-28:
Yes, this is an architectural shortcoming of our qtgui blocks. We really must stop using variables directly, and make these setters use message passing under the hood, and make the message handlers use the right method to do the emitting of signals in the right thread

SumiSpam, 2025-09-06:
Really thanks to all!

You have solved my problem and opened up other possibilities that I was looking for, like control a QT GUI Range with a variable, or modify values in a QT GUI Vector Sink (or even its line and marker style).

The attached example illustrates some of the above.
### Parts
- get access to the top_block in GRC, creating a new Variable Block named Self
[image]

- use the function eval() to get attributes of objects
[image]

- use an extended version of the Jeija' workaround to pass values to the Qt main thread
[image]

### Screenshots
When 'enable':
- 'var_local' controls a 'var' in a QT GUI hier block
- the line style of the Vector Sink changes, and you can choose a marker style.
- all without "Cannot send..." messages

[image]
[image]

### File
[test_top_block.zip](https://github.com/user-attachments/files/22189845/test_top_block.zip)

Zip file includes a README.txt
Just an example, use at your own risk!
Sure, there are best approaches to do this (I don't know how to build an OOT Block yet).