gnuradio/gnuradio #4702: Message Pair to Var is not working with messages sent by Embedded Python Block

eggelene, 2021-06-01:
When a python blocks send Messages to a `Message to Var block`, the var is always set to None. Messages sent by the `Variable to Message` and `QT GUI Toggle Button` are handled correctly ( meaning the right value is shown in the display)

GNURadio 3.9.1.0 (Python 3.8.5) is running on Ubuntu 20.4 ( fully updated ) on Core i7.

Please find the graph, code and low below.

[image]
[image]

 
Generating: '/home/rfstreamer/Documents/GNURadio/message.py'

Executing: /usr/bin/python3 -u /home/rfstreamer/Documents/GNURadio/message.py

```
.0..4096..6144..7168..7680..7936..8064..8128..8160..8176..8184..8188..8190..8191..12287..14335..15359..15871..16127..16255..16319..16351..16367..16375..16379..16381..16382..20478..22526..23550..24062..24318..24446..24510..24542..24558..24566..24570..24572..24573..28669..30717..31741.******* MESSAGE DEBUG PRINT ********
((test . 123))
************************************
.0..256..384..448..480..496..504..508..510..511..4607..6655..7679..8191..8447..8575..8639..8671..8687..8695..8699..8701..8702..12798..14846..15870..16382..16638..16766..16830..16862..16878..16886..16890..16892..16893..20989..23037..24061..24573..24829..24957..25021..25053..25069..25077..25081..25083..25084..29180..31228.******* MESSAGE DEBUG PRINT ********
((test . 124))
************************************
.0..512..768..896..960..992..1008..1016..1020..1022..1023..5119..7167..8191..8703..8959..9087..9151..9183..9199..9207..9211..9213..9214..13310..15358..16382..16894..17150..17278..17342..17374..17390..17398..17402..17404..17405..21501..23549..24573..25085..25341..25469..25533..25565..25581..25589..25593..25595..25596..29692..31740.******* MESSAGE DEBUG PRINT ********
((test . 125))
************************************
.0..512..768..896..960..992..1008..1016..1020..1022..1023..5119..7167..8191..8703..8959..9087..9151..9183..9199..9207..9211..9213..9214..13310..15358..16382..16894..17150..17278..17342..17374..17390..17398..17402..17404..17405..21501..23549..24573..25085..25341..25469..25533..25565..25581..25589..25593..25595..25596..29692..31740.******* MESSAGE DEBUG PRINT ********
((test . 126))
************************************
.0..512..768..896..960..992..1008..1016..1020..1022..1023..5119..7167..8191..8703..8959..9087..9151..9183..9199..9207..9211..9213..9214..13310..15358..16382..16894..17150..17278..17342..17374..17390..17398..17402..17404..17405..21501..23549..24573..25085..25341..25469..25533..25565..25581..25589..25593..25595..25596..29692..31740.******* MESSAGE DEBUG PRINT ********
((test . 127))
************************************
.0..512..768..896..960..992..1008..1016..1020..1022.******* MESSAGE DEBUG PRINT ********
(test . 50)
************************************
.1023..5119..7167..8191..8703..8959..9087..9151..9183..9199..9207..9211..9213..9214..13310..15358..16382..16894..17150..17278..17342..17374..17390..17398..17402..17404..17405..21501..23549..24573..25085..25341..25469..25533..25565..25581..25589..25593..25595..25596..29692..31740.******* MESSAGE DEBUG PRINT ********
((test . 128))
************************************
.0..512..768..896..960..992..1008..1016..1020..1022..1023..5119..7167..8191..8703..8959..9087..9151..9183..9199..9207..9211..9213..9214..13310..15358..16382..16894..17150..17278..17342..17374..17390..17398..17402..17404..17405..21501..23549..24573..25085..25341..25469..25533..25565..25581..25589..25593..25595..25596..29692..31740.******* MESSAGE DEBUG PRINT ********
((test . 129))
************************************
.0..512..768..896..960..992..1008..1016..1020..1022..1023..5119..7167..8191..8703..8959..9087..9151..9183..9199..9207..9211..9213..9214..13310..15358..16382..16894..17150..17278..17342..17374..17390..17398..17402..17404..17405..21501..23549..24573..25085..25341..25469..25533..25565..25581..25589..25593..25595..25596..29692..31740.******* MESSAGE DEBUG PRINT ********
((test . 130))
************************************
.0..512..768..896..960..992..1008..1016..1020..1022..1023..5119..7167..8191..8703..8959..9087..9151..9183..9199..9207..9211..9213..9214..13310..15358..16382..16894..17150..17278..17342..17374..17390..17398..17402..17404..17405.******* MESSAGE DEBUG PRINT ********
(test . 60)
************************************
.21501..23549..24573..25085..25341..25469..25533..25565..25581..25589..25593..25595..25596..29692..31740.******* MESSAGE DEBUG PRINT ********
((test . 131))
************************************
.0..512..768..896..960..992..1008..1016..1020..1022..1023..5119..7167..8191..8703..8959..9087..9151..9183..9199..9207..9211..9213..9214..13310..15358..16382..16894..17150..17278..17342..17374..17390..17398..17402..17404..17405..21501..23549..24573..25085..25341..25469..25533..25565..25581..25589..25593..25595..25596..29692..31740.******* MESSAGE DEBUG PRINT ********
((test . 132))
************************************
.0..512..768..896..960..992..1008..1016..1020..1022..1023..5119..7167..8191..8703..8959..9087..9151..9183..9199..9207..9211..9213..9214..13310..15358..16382..16894..17150..17278..17342..17374..17390..17398..17402..17404..17405..21501..23549..24573..25085..25341..25469..25533..25565..25581..25589..25593..25595..25596..29692..31740.******* MESSAGE DEBUG PRINT ********
((test . 133))
************************************
.0..512..768..896..960..992..1008..1016..1020..1022..1023.
>>> Done

```
```
"""
Embedded Python Blocks:

Each time this file is saved, GRC will instantiate the first class it finds
to get ports and parameters of your block. The arguments to __init__  will
be the parameters. All of them are required to have default values!
"""

import pmt
import numpy as np
from gnuradio import gr

class blk(gr.sync_block):  # other base classes are basic_block, decim_block, interp_block
    """Embedded Python Block example - a simple multiply const"""

    def __init__(self, example_param=1.0):  # only default arguments here
        """arguments to this function show up as parameters in GRC"""
        gr.sync_block.__init__(
            self,
            name='Embedded Python Block',   # will show up in GRC
            in_sig=[np.complex64],
            out_sig=[np.complex64]
        )
        self.message_port_register_out(pmt.intern('msg_out'))
        # if an attribute with the same name as a parameter is found,
        # a callback is registered (properties work, too).
        self.example_param = example_param
        self.sendMessage=0
        self.sendMessageCount=123

    def work(self, input_items, output_items):
    
        print (".%d." % (self.sendMessage),end="")
        self.sendMessage=self.sendMessage+len(output_items[0])
        if (self.sendMessage>=32000):
            self.sendMessage=0
            
            key0 = pmt.intern("test")
            val0 = pmt.from_long(self.sendMessageCount)
            
            msg = pmt.make_dict()
            msg = pmt.dict_add(msg, key0, val0)
            
            self.message_port_pub(pmt.intern('msg_out'), msg );
            
            self.sendMessageCount=self.sendMessageCount+1
            
        """example: multiply with constant"""
        output_items[0][:] = input_items[0] * self.example_param
        return len(output_items[0])
```

jacobagilbert, 2021-06-01:
Does the reference flowgraph `gr_blocks/examples/msg_to_var.grc` work for you?

eggelene, 2021-06-01:
I just checked. This works too.

I tried all kind of combinations with different blocks ( they all work), but I cannot get it to work Embedded Python block. The strange thing is that the debug window receives the messages correctly ( see log ). e.g. The message created by the var to message `(test . 60)` is shown correctly in the window. 

[image]

jacobagilbert, 2021-06-01:
It looks like you are providing that block a dictionary when the `Documentation` tab suggests it wants a pair.

Try changing

```
            msg = pmt.make_dict()
            msg = pmt.dict_add(msg, key0, val0)
            
            self.message_port_pub(pmt.intern('msg_out'), msg );
```

to 

```
            self.message_port_pub(pmt.intern('msg_out'), pmt.cons(key0, val0);
```

eggelene, 2021-06-01:
Thanks for you feedback. I checked, and can confirm that this modification solves the issue.

You are right, when I re-read the documentation it indicates that.  Is it possible to propose updates to the documentation here ?

jacobagilbert, 2021-06-01:
You can propose updates, certainly.

jacobagilbert, 2021-06-02:
@eggelene I added some info to the docstrings in the PR linked above. Let me know if there is other information you feel would help in there.

PR #4704 (2021-06-07): gr-blocks: improvements to input handling in var/msg blocks
Fixes #4702 where a user was feeding a dictionary into the msg_to_var
block which was then setting the variable to 'None' causing unexpected
behavior. This improves input sanitization for both blocks and adds
some extra info to the docstrings.
closed 2021-06-07 by mormj