gnuradio/gnuradio #4703: synchronization of the update of processing ( filters) blocks variables

eggelene, 2021-06-01:
Based on discussion with issue  #4646 this issue is created.

> Updates to variables (including taps) take place "as soon as possible". The exact timing and order of updates is not specified.
So when a python blocks updates e.g. low pass frequency and high pass frequency this means that some samples could be processed with only updating e.g. low pass frequency. 

It would help if some kind synchronization will be defined within GNUradio. My target would be that parameters can be updated atomically on a define timestamp ( or other means)  

Example:
When you have a band-pass and band-suppress in your graph ( to split the signal) with the intention to process one band with different blocks and merge it afterward, it would be really helpful is e.g. the high pass frequency and low pass frequency can be changed such that they are applied on the same sample ( even when blocks before or after the band filters introduce a delay). By having a perfect synchronization, it would be possible to change the band filters and have no glitches in the signal.

willcode, 2021-06-02:
This is a potentially useful, but very major feature. Tags can already help with part of this, but most blocks are not set up to respond in this way. And, as we discussed (quote above), GNU Radio's messaging and streaming paths are not sync'd.

jacobagilbert, 2021-06-02:
I can see some niche use cases for this, in fact I developed a clone of the in-tree freq_xlating_fir_filter block that has the ability to set the frequency on a sample-accurate basis [here](https://github.com/sandialabs/gr-timing_utils/blob/master/lib/timed_freq_xlating_fir_impl.cc#L281) for a similar general objective some years ago. This might be a good reference if you are interested in how you might be able to customize GR to do this and use stream tags to modify several blocks simultaneously; this could pretty easily be extended to set the filter taps and do what you want (you don't need to use the decimation or frequency translation available in that block).

In terms of a general runtime-level concept of synchronization, that really does not generalize well into the current runtime environment, and given the very specialized use cases I suspect this will be something left to custom blocks similar to what was linked above.

eggelene, 2021-06-03:
Based on what jacoba made. Would the following be an idea : use the normal variable system to change filter parameters. But add one setting to the GNUradio blocks that apply these settings not "as soon as possible, but to trigger this on a defined stream tags  ?

willcode, 2021-06-03:
Setup and trigger is a good idea in general. Hard to implement this well in GNU Radio under the current architecture. You would want to be able to define a "next state" for a block, which might include multiple settings. This could be a GR4.0 goal. We'd have to be careful modifying specific blocks to do something really specific. You could potentially subclass some of the filter blocks to have this behavior.