gnuradio/gnuradio #4714: Segfault on FreeBSD

onno182, 2021-06-04:
I complied GNU Radio on FreeBSD succesfully, but get a segfault when running a simple .grc with a multiply block. 

The .grc works fine on Pentoo.

`Traceback (most recent call last):
  File "lesson1.py", line 165, in <module>
    main()
  File "lesson1.py", line 143, in main
    tb = top_block_cls()
  File "lesson1.py", line 112, in __init__
    self.connect((self.analog_sig_source_x_0_0, 0), (self.blocks_multiply_xx_0, 1))
  File "/usr/local/gnuradio/lib/python3.7/site-packages/gnuradio/gr/hier_block2.py", line 37, in wrapped
    func(self, src, src_port, dst, dst_port)
  File "/usr/local/gnuradio/lib/python3.7/site-packages/gnuradio/gr/hier_block2.py", line 100, in connect
    self.primitive_connect(*args)
ValueError: port number 1 exceeds max of 0
Segmentation fault (core dumped)
`

willcode, 2021-06-04:
You should certainly be able to connect to port 1 on a multiply block without a core dump! FreeBSD is not a common platform for GNU Radio, and I don't see much related on Google in the last few years, so it may be a while before anyone else tries this out.

marcusmueller, 2021-06-04:
This would look like a linker confusion (the backtrace showing you're in an unrelated function to the place of the segfault): @onno182 are you sure there's no older GNU Radio headers still lying around on your system? 

I'm not a FreeBSD user, but if you tell us whether there are any special steps necessary to compile GNU Radio on a fresh FreeBSD VM, someone might come around to testing it!

onno182, 2021-06-05:
I have reinstalled FreeBSD from scratch and rebuilt GNU Radio and Volk from source.  The problem unfortunately remains.

The only special step for compiling with FreeBSD is to modify the CPU header file of Volk as it does not recognise FreeBSD as a supported operating system.  The solution is to make the preprocessor handle it the same as Linux.  Other than that all dependencies are available in the FreeBSD package system.   The FindQWT cmake file also has to be modified as FreeBSD places the include and library files of QWT in different directories than where cmake looks.

Of course FreeBSD also needs the proc filesystem enabled, but this will already be done when running a GUI like Gnome or KDE anyway.

There is a GNU Radio version in the package system, but unfortunately this is broken as it tries to uninstall Gnome3.

vishwin, 2021-06-05:
I disabled `cpu_features` from libvolk completely, you may want to try that.

(side note: I'm working on a ports update to 3.9)

marcusmueller, 2021-06-06:
> The only special step for compiling with FreeBSD is to modify the CPU header file of Volk as it does not recognise FreeBSD as a supported operating system. The solution is to make the preprocessor handle it the same as Linux.

Care to share that modification? Might be worth an upstreaming :)

vishwin, 2021-06-06:
I tried to do that, but it's not foolproof. Our sysctls are different from Linux and macOS/Darwin, among other things, so the whole point of `cpu_features` in libvolk may not operate properly.

vishwin, 2021-06-06:
Also should mention that FreeBSD's `procfs` is completely different than Linux `procfs`. A FreeBSD binary (compared to a Linux binary running via the Linuxulator) should not be reading `cpuinfo` from our `linprocfs`.

onno182, 2021-06-06:
@vishwin The error remains with cpu_features disabled. On the other hand with it activated, volk_profile did run successfully.

onno182, 2021-06-06:
@marcusmueller the patches to compile on FreeBSD are the following. Of course I do not know if the library functions correctly given the previous comments, although disable cpu_features does not fix the problem.

--- ../cpu_features/include/cpu_features_macros.h.orig	2021-06-06 13:08:05.165204000 +0200
+++ ../cpu_features/include/cpu_features_macros.h	2021-06-06 13:14:18.621659000 +0200
@@ -12,6 +12,8 @@
 // See the License for the specific language governing permissions and
 // limitations under the License.
 
+#include <sys/param.h>
+
 #ifndef CPU_FEATURES_INCLUDE_CPU_FEATURES_MACROS_H_
 #define CPU_FEATURES_INCLUDE_CPU_FEATURES_MACROS_H_
 
@@ -81,6 +83,10 @@
 
 #if (defined(__apple__) || defined(__APPLE__) || defined(__MACH__))
 #define CPU_FEATURES_OS_DARWIN
+#endif
+
+#if (defined(__FreeBSD__))
+#define CPU_FEATURES_OS_FREEBSD
 #endif
 
 ////////////////////////////////////////////////////////////////////////////////

--- ../cpu_features/src/cpuinfo_x86.c.orig	2021-06-06 13:03:42.821434000 +0200
+++ ../cpu_features/src/cpuinfo_x86.c	2021-06-06 13:15:05.460556000 +0200
@@ -106,6 +106,10 @@
 #error "Darwin needs support for sysctlbyname"
 #endif
 #include <sys/sysctl.h>
+#elif defined(CPU_FEATURES_OS_FREEBSD) 
+#include "internal/filesystem.h"        // Needed to parse /proc/cpuinfo
+#include "internal/stack_line_reader.h"  // Needed to parse /proc/cpuinfo
+#include "internal/string_view.h"        // Needed to parse /proc/cpuinfo
 #else
 #error "Unsupported OS"
 #endif  // CPU_FEATURES_OS
@@ -1240,6 +1244,31 @@
   features->sse4_1 = GetDarwinSysCtlByName("hw.optional.sse4_1");
   features->sse4_2 = GetDarwinSysCtlByName("hw.optional.sse4_2");
 #elif defined(CPU_FEATURES_OS_LINUX_OR_ANDROID)
+  // Handling Linux platform through /proc/cpuinfo.
+  const int fd = CpuFeatures_OpenFile("/proc/cpuinfo");
+  if (fd >= 0) {
+    StackLineReader reader;
+    StackLineReader_Initialize(&reader, fd);
+    for (;;) {
+      const LineResult result = StackLineReader_NextLine(&reader);
+      const StringView line = result.line;
+      StringView key, value;
+      if (CpuFeatures_StringView_GetAttributeKeyValue(line, &key, &value)) {
+        if (CpuFeatures_StringView_IsEquals(key, str("flags"))) {
+          features->sse = CpuFeatures_StringView_HasWord(value, "sse");
+          features->sse2 = CpuFeatures_StringView_HasWord(value, "sse2");
+          features->sse3 = CpuFeatures_StringView_HasWord(value, "sse3");
+          features->ssse3 = CpuFeatures_StringView_HasWord(value, "ssse3");
+          features->sse4_1 = CpuFeatures_StringView_HasWord(value, "sse4_1");
+          features->sse4_2 = CpuFeatures_StringView_HasWord(value, "sse4_2");
+          break;
+        }
+      }
+      if (result.eof) break;
+    }
+    CpuFeatures_CloseFile(fd);
+  }
+#elif defined(CPU_FEATURES_OS_FREEBSD)
   // Handling Linux platform through /proc/cpuinfo.
   const int fd = CpuFeatures_OpenFile("/proc/cpuinfo");
   if (fd >= 0) {

`

onno182, 2021-06-06:
@marcusmueller sorry for the messy post (cant upload patch files here, but see .txt file for patch)

marcusmueller, 2021-06-06:
@onno182 absolutely nothing to be sorry about!

I'm taking the freedom to transfer this to the VOLK issue tracker, for now. (We can always transfer back, or just open more issues here, and I feel that this is first fixing VOLK, then we can come back to GNU Radio)