gnuradio/gnuradio #4727: mako version check is wrong

vishwin, 2021-06-05:
During cmake configure, when running `GR_PYTHON_CHECK_MODULE("Mako >= ${GR_MAKO_MIN_VERSION}" mako "mako.__version__ >= '${GR_MAKO_MIN_VERSION}'" MAKO_FOUND)` on a system with mako 1.0.14 installed, it reports `MAKO_FOUND = FALSE`. On v3.9.1.0, `GR_MAKO_MIN_VERSION` is 1.0.7. This is due to the string comparison:
```
>>> import mako
>>> mako.__version__
'1.0.14'
>>> mako.__version__ >= '1.0.7'
False
>>>
```
However, on systems with mako 1.1.0 and above, the check is correct.

marcusmueller, 2021-06-06:
does 

    "tuple(int(s) for s in mako.__version__.split(".")) >= tuple(int(s) for s in '${GR_MAKO_MIN_VERSION}'.split("."))"

solve that issue?

willcode, 2021-06-06:
This cmake line is from 2017, so it would affect all current branches.

vishwin, 2021-06-06:
Works:
```
--- gr-utils/CMakeLists.txt.orig        2021-03-22 22:37:12 UTC
+++ gr-utils/CMakeLists.txt
@@ -10,7 +10,7 @@
 ########################################################################
 include(GrPython)

-GR_PYTHON_CHECK_MODULE("Mako >= ${GR_MAKO_MIN_VERSION}" mako "mako.__version__ >= '${GR_MAKO_MIN_VERSION}'" MAKO_FOUND)
+GR_PYTHON_CHECK_MODULE("Mako >= ${GR_MAKO_MIN_VERSION}" mako "tuple(int(s) for s in mako.__version__.split('.')) >= tuple(int(s) for s in '${GR_MAKO_MIN_VERSION}'.split('.'))" MAKO_FOUND)

 GR_PYTHON_CHECK_MODULE_RAW(
   "click"
```
Will also need to apply a similar change for grc:
```
--- grc/CMakeLists.txt.orig     2021-03-22 22:37:12 UTC
+++ grc/CMakeLists.txt
@@ -20,7 +20,7 @@ GR_PYTHON_CHECK_MODULE_RAW(

 GR_PYTHON_CHECK_MODULE_RAW(
     "mako >= ${GR_MAKO_MIN_VERSION}"
-    "import mako; assert mako.__version__ >= '${GR_MAKO_MIN_VERSION}'"
+    "import mako; assert tuple(int(s) for s in mako.__version__.split('.')) >= tuple(int(s) for s in '${GR_MAKO_MIN_VERSION}'.split('.'))"
     MAKO_FOUND
 )

```
Column width is a bit long but that's a style nit.

marcusmueller, 2021-06-06:
Thanks for checking! Do you want to open that pull request, @vishwin, or should I? All we'd need you to do is use `git commit -s`, to sign-off the commit (i.e. implying that you're OK with the code being contributed, and not using e.g. your employer's intellectual property without consent).

willcode, 2021-06-06:
What targets are using 1.0.1X at the moment, i.e., is it urgent to get this into next week's releases or can it wait 3 months?

vishwin, 2021-06-06:
FreeBSD ports still has 1.0.14. We're a bit conservative when it comes to updating pypi/cheeseshop packages in our tree, not least because they're harder to test compared to compiled software.

willcode, 2021-06-06:
How about using built-in functions:
```
from distutils.version import StrictVersion
StrictVersion("1.0.14") > StrictVersion("1.0.7")
```

closed 2021-07-19 by mormj