gnuradio/gnuradio #4728: zeromq: Setsockopt is deprecated

haakov, 2021-06-05:
Compiling on Arch Linux, Zeromq version 4.3.4-1.
```
gr-zeromq/lib/base_impl.cc:54:24: warning: ‘void zmq::detail::socket_base::getsockopt(int, void*, size_t*) const’ is deprecated: from 4.7.0, use `get` taking option from zmq::sockopt [-Wdeprecated-declarations]
   54 |     d_socket.getsockopt(ZMQ_LAST_ENDPOINT, addr, &addr_len);
      |     ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/hkon/gr_start/gr-zeromq/lib/zmq_common_impl.h:14,
                 from /home/hkon/gr_start/gr-zeromq/lib/base_impl.h:14,
                 from /home/hkon/gr_start/gr-zeromq/lib/base_impl.cc:15:
/usr/include/zmq.hpp:1670:10: note: declared here
 1670 |     void getsockopt(int option_, void *optval_, size_t *optvallen_) const
      |          ^~~~~~~~~~
```
Full warning log below

haakov, 2021-06-06:
```
/home/hkon/gr_start/gr-zeromq/lib/base_impl.cc: In member function ‘std::string gr::zeromq::base_impl::last_endpoint()’:
/home/hkon/gr_start/gr-zeromq/lib/base_impl.cc:54:24: warning: ‘void zmq::detail::socket_base::getsockopt(int, void*, size_t*) const’ is deprecated: from 4.7.0, use `get` taking option from zmq::sockopt [-Wdeprecated-declarations]
   54 |     d_socket.getsockopt(ZMQ_LAST_ENDPOINT, addr, &addr_len);
      |     ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/hkon/gr_start/gr-zeromq/lib/zmq_common_impl.h:14,
                 from /home/hkon/gr_start/gr-zeromq/lib/base_impl.h:14,
                 from /home/hkon/gr_start/gr-zeromq/lib/base_impl.cc:15:
/usr/include/zmq.hpp:1670:10: note: declared here
 1670 |     void getsockopt(int option_, void *optval_, size_t *optvallen_) const
      |          ^~~~~~~~~~
/home/hkon/gr_start/gr-zeromq/lib/base_impl.cc: In constructor ‘gr::zeromq::base_sink_impl::base_sink_impl(int, size_t, size_t, char*, int, bool, int, const string&)’:
/home/hkon/gr_start/gr-zeromq/lib/base_impl.cc:72:28: warning: ‘void zmq::detail::socket_base::setsockopt(int, const void*, size_t)’ is deprecated: from 4.7.0, use `set` taking option from zmq::sockopt [-Wdeprecated-declarations]
   72 |         d_socket.setsockopt(ZMQ_SNDHWM, &hwm, sizeof(hwm));
      |         ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/hkon/gr_start/gr-zeromq/lib/zmq_common_impl.h:14,
                 from /home/hkon/gr_start/gr-zeromq/lib/base_impl.h:14,
                 from /home/hkon/gr_start/gr-zeromq/lib/base_impl.cc:15:
/usr/include/zmq.hpp:1662:10: note: declared here
 1662 |     void setsockopt(int option_, const void *optval_, size_t optvallen_)
      |          ^~~~~~~~~~
/home/hkon/gr_start/gr-zeromq/lib/base_impl.cc:80:24: warning: ‘void zmq::detail::socket_base::setsockopt(int, const void*, size_t)’ is deprecated: from 4.7.0, use `set` taking option from zmq::sockopt [-Wdeprecated-declarations]
   80 |     d_socket.setsockopt(ZMQ_LINGER, &LINGER_DEFAULT, sizeof(LINGER_DEFAULT));
      |     ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/hkon/gr_start/gr-zeromq/lib/zmq_common_impl.h:14,
                 from /home/hkon/gr_start/gr-zeromq/lib/base_impl.h:14,
                 from /home/hkon/gr_start/gr-zeromq/lib/base_impl.cc:15:
/usr/include/zmq.hpp:1662:10: note: declared here
 1662 |     void setsockopt(int option_, const void *optval_, size_t optvallen_)
      |          ^~~~~~~~~~
/home/hkon/gr_start/gr-zeromq/lib/base_impl.cc: In constructor ‘gr::zeromq::base_source_impl::base_source_impl(int, size_t, size_t, char*, int, bool, int, const string&)’:
/home/hkon/gr_start/gr-zeromq/lib/base_impl.cc:146:28: warning: ‘void zmq::detail::socket_base::setsockopt(int, const void*, size_t)’ is deprecated: from 4.7.0, use `set` taking option from zmq::sockopt [-Wdeprecated-declarations]
  146 |         d_socket.setsockopt(ZMQ_RCVHWM, &hwm, sizeof(hwm));
      |         ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/hkon/gr_start/gr-zeromq/lib/zmq_common_impl.h:14,
                 from /home/hkon/gr_start/gr-zeromq/lib/base_impl.h:14,
                 from /home/hkon/gr_start/gr-zeromq/lib/base_impl.cc:15:
/usr/include/zmq.hpp:1662:10: note: declared here
 1662 |     void setsockopt(int option_, const void *optval_, size_t optvallen_)
      |          ^~~~~~~~~~
/home/hkon/gr_start/gr-zeromq/lib/base_impl.cc:154:24: warning: ‘void zmq::detail::socket_base::setsockopt(int, const void*, size_t)’ is deprecated: from 4.7.0, use `set` taking option from zmq::sockopt [-Wdeprecated-declarations]
  154 |     d_socket.setsockopt(ZMQ_LINGER, &LINGER_DEFAULT, sizeof(LINGER_DEFAULT));
      |     ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/hkon/gr_start/gr-zeromq/lib/zmq_common_impl.h:14,
                 from /home/hkon/gr_start/gr-zeromq/lib/base_impl.h:14,
                 from /home/hkon/gr_start/gr-zeromq/lib/base_impl.cc:15:
/usr/include/zmq.hpp:1662:10: note: declared here
 1662 |     void setsockopt(int option_, const void *optval_, size_t optvallen_)
      |          ^~~~~~~~~~
/home/hkon/gr_start/gr-zeromq/lib/base_impl.cc: In member function ‘bool gr::zeromq::base_source_impl::load_message(bool)’:
/home/hkon/gr_start/gr-zeromq/lib/base_impl.cc:203:24: warning: ‘void zmq::detail::socket_base::getsockopt(int, void*, size_t*) const’ is deprecated: from 4.7.0, use `get` taking option from zmq::sockopt [-Wdeprecated-declarations]
  203 |     d_socket.getsockopt(ZMQ_RCVMORE, &more, &more_len);
      |     ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/hkon/gr_start/gr-zeromq/lib/zmq_common_impl.h:14,
                 from /home/hkon/gr_start/gr-zeromq/lib/base_impl.h:14,
                 from /home/hkon/gr_start/gr-zeromq/lib/base_impl.cc:15:
/usr/include/zmq.hpp:1670:10: note: declared here
 1670 |     void getsockopt(int option_, void *optval_, size_t *optvallen_) const
      |          ^~~~~~~~~~
/home/hkon/gr_start/gr-zeromq/lib/base_impl.cc:228:28: warning: ‘void zmq::detail::socket_base::getsockopt(int, void*, size_t*) const’ is deprecated: from 4.7.0, use `get` taking option from zmq::sockopt [-Wdeprecated-declarations]
  228 |         d_socket.getsockopt(ZMQ_RCVMORE, &is_multipart, &more_len);
[... cut ...]
   48 |     d_socket.setsockopt(ZMQ_LINGER, &time, sizeof(time));
      |     ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/hkon/gr_start/gr-zeromq/lib/zmq_common_impl.h:14,
                 from /home/hkon/gr_start/gr-zeromq/lib/req_msg_source_impl.h:14,
                 from /home/hkon/gr_start/gr-zeromq/lib/req_msg_source_impl.cc:15:
/usr/include/zmq.hpp:1662:10: note: declared here
 1662 |     void setsockopt(int option_, const void *optval_, size_t optvallen_)
      |          ^~~~~~~~~~
```

marcusmueller, 2021-06-06:
Yeah, looked into that last month. Small problem is that the proposed replacement of these deprecated functions is not yet there in the most versions of zeromq that we support. So, that would mean `#ifdef`'ed code paths for old and new zeromq.

But it's nothing we'll be able to avoid in the long run.

haakov, 2022-07-15:
Duplicate of https://github.com/gnuradio/gnuradio/issues/5982, closing.

closed 2022-07-15 by haakov